fritzing/fritzing-app #3310: Core Parts Missing in 0.9.3b

granitepenguin, 2017-02-03:
OS: macOS 10.11.6
Fritzing version: 0.9.3b

Just updated from 0.9.2 and now when Fritzing starts, I get the following error when loading Core Parts:
[image]

Tried deleting my fritzing directory and starting again, but problem still persists.

VeitSanner, 2017-03-07:
@granitepenguin, do you experience the same issues as detailed in #3293?

granitepenguin, 2017-03-07:
basically, yes.  I get the same "Parts > Regenerate parts database..." suggestion but I can't do that because it's greyed out in the menu, so I never get to the "Unable to open temporary file" dialog.

granitepenguin, 2017-03-07:
If I open a new project, I get the Regenerate Parts Database dialog, and it does do the same thing, so this looks like it is a duplicate of #3293

VeitSanner, 2017-03-09:
@granitepenguin, thanks for the confirmation.

@aknoerig, is there a chance to close this issue and use #3293 for als further discussions?

granitepenguin, 2017-03-09:
FWIW, I would be fine with that.

aknoerig, 2017-12-18:
Closed as duplicate of the above mentioned.

closed 2017-12-18 by aknoerig