fritzing/fritzing-app #3322: Inability to upload code to the arduino

np511, 2017-04-05:
After a little research, I found that uploading code to an Arduino isn't functional.

Why is that? I know for a fact it's possible on all platforms, I've done some tinkering with it.

Arduino comes with (or, on Linux distros such as Debian and Ubuntu, depends on) a program called `avrdude` which actually uploads the file. Compiling is a bit more tricky on Linux distros, but when Arduino is installed through a downloaded zip, it comes with a program that can be invoked easily.

KjellMorgenstern, 2019-06-29:
It would be great to be able to publish code + circuit in one project, as those are often tightly coupled. 
But as soon as you start to write a bit more code, even the Arduino IDE might be to simplified. So, maybe it would be better to enable Fritzing to refer the code, tag it or bundle a copy of the code (and even the binary) in your projects. Also Arduino has been decouple their IDE frontend and backend recently...

deshipu, 2019-06-30:
I am a bit skeptical about the usefulness of the IDE in Fritzing. Programmers usually have their own favorite tools (not just the IDE itself, but a whole bunch of linters, formatters, indexers, documentation generators and so on), and it would make the scope of the project explode to support it all. Bundling the code with the project also doesn't sound so good, since you usually only develop the hardware until you send it to fabrication, and you work on the software once you receive it, and keep on improving it long past that version of the design has been finalized. But maybe my use cases are not the common ones.

I wonder how many users would actually miss that tab if it was removed — the fact that nobody noticed earlier that it no longer works probably suggests something.

KjellMorgenstern, 2019-07-01:
@deshipu Yes, a full blown IDE is just to much, and right now I find it just distracting.

But how would the workflow for an Open Source Hardware project look like? Or if you want a very fast development cycle for your hardware? Our just a long lived project?

In those cases you have many versions of the hardware floating around, and probably even more software iterations and forks. That might still be a bit academic. But wouldn't you want some kind of version management (e.g. a git repo, with forks and so on), that manages both, hard and software? Not sure how that would look like, certainly not the fritzing code editor in its current form. But I think that was one of the motives why an editor was added in the first place.

deshipu, 2019-07-01:
It is a big problem, and I don't think there is one solution that works for everyone. I recently stumbled into that with one of my projects, and this is the repository structure I came up with: https://github.com/pewpew-game/ — with separate repositories that I can cherry-pick between. There are five or six different hardware versions, libraries for each of those, and the actual games. Each of those needs a separate release cycle, etc. I don't think anything that could be quickly added to Fritzing would help with that.

aknoerig, 2019-07-01:
Just for reference, the main motivations for the code editor were to

1. make it easier for total noobs to get started with Arduino & co - and also make it easy to switch between microcontrollers (it also supports the somewhat exotic Picaxe platform).
2. for advanced users, make it easier to store your code along with the matching version of your circuit
3. eventually enable cross-linking between code and circuit elements, as in `LED l1; l1.on()` etc.

Regarding the broken code, the current implementation is based on an [outdated version of the Arduino command line](https://github.com/arduino/Arduino/blob/ide-1.5.x/build/shared/manpage.adoc).
It would need to get updated to the more future-proof evolution of it, though I'm not sure which that is... There's at least [arduino-builder](https://github.com/arduino/arduino-builder) and [arduino-cli](https://github.com/arduino/arduino-cli).

There's a little more info on the implementation in the [fritzing wiki here](https://github.com/fritzing/fritzing-app/wiki/2.3-Adding-a-platform-to-Code-view).

WimberleyTech, 2019-07-01:
I think only #2 is useful.

pixelherodev, 2019-07-02:
Personally, I can see 3 being useful also, but as long as two works, I'm happy

KjellMorgenstern, 2019-07-04:
Thanks for all the input! So I think the code editor will stay. And it seems to work fine, I just tried it with the Linux pre-release version (https://github.com/fritzing/fritzing-app/releases/tag/CD-268)

I used the 'arduino' executable, not 'avrdude'.

What I did to test this:
0. I have arduino 1.8.8 installed with umake on ubuntu 18.04 budgie.
1. Delete my $HOME/Documents/Fritzing directory (not neccessary, just wanted to make sure to have no side effects from it)
2. unpack the linux release, open the directory, directly start Fritzing (without running the installer)
3. Go to edit -> preferences -> code view, enter /home/<username>/.local/share/umake/bin/arduino
Thats where it was installed by umake
4. Connect an Elegoo Mega2560 R3 (I verified it has a correct bootloader by testing it with Arduino IDE first)
5. The top menu somehow does not show Board types (bug!) but via the red bar at the bottom I can set Platform, Board and Port. 
6 Enter some .ino code (blink), upload, works fine. Change blink frequency, upload again, works fine with new blink frequency.

KjellMorgenstern, 2019-07-07:
Closing this issue, since uploading is well possible. PR #3501 contains a fix for the strange behavior of the Menu observed during investigating this issue.

closed 2019-07-07 by KjellMorgenstern