fritzing/fritzing-app #3324: Fritzing crashes for no apparent reason.

juniorRubyist, 2017-04-05:
Whenever I open Fritzing, it opens OK, but after ~30 seconds, it will crash. I am running a Mac with OS X 10.7 (old, I know) that is a new install.
Here is an error trace-back:
```
Process:         Fritzing [961]
Path:            /Applications/Fritzing.app/Contents/MacOS/Fritzing
Identifier:      org.fritzing.Fritzing
Version:         0.9.3 (bda4e)
Code Type:       X86-64 (Native)
Parent Process:  launchd [253]

Date/Time:       2017-04-05 08:43:46.910 -0700
OS Version:      Mac OS X 10.7 (11A511)
Report Version:  9

Interval Since Last Report:          1975 sec
Crashes Since Last Report:           6
Per-App Interval Since Last Report:  110 sec
Per-App Crashes Since Last Report:   6
Anonymous UUID:                      7671FB67-E607-4FC3-9704-FE1BB577FBD8

Crashed Thread:  0  Dispatch queue: com.apple.main-thread

Exception Type:  EXC_BREAKPOINT (SIGTRAP)
Exception Codes: 0x0000000000000002, 0x0000000000000000

Application Specific Information:
objc[961]: garbage collection is OFF

Dyld Error Message:
  Symbol not found: _SSLCreateContext
  Referenced from: /Applications/Fritzing.app/Contents/MacOS/../Frameworks/libgit2.23.dylib
  Expected in: /System/Library/Frameworks/Security.framework/Versions/A/Security

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   dyld                          	0x00007fff6c3f706d dyld_fatal_error + 1
1   dyld                          	0x00007fff6c3fa918 dyld::fastBindLazySymbol(ImageLoader**, unsigned long) + 128
2   libdyld.dylib                 	0x00007fff8fe35716 dyld_stub_binder_ + 13
3   ???                           	0x000000010d33b018 0 + 4516458520
4   libgit2.23.dylib              	0x000000010d2ff0d5 0x10d226000 + 889045
5   libgit2.23.dylib              	0x000000010d30f4fa 0x10d226000 + 955642
6   libgit2.23.dylib              	0x000000010d30dffc 0x10d226000 + 950268
7   libgit2.23.dylib              	0x000000010d3134d9 0x10d226000 + 971993
8   libgit2.23.dylib              	0x000000010d2d8ae8 git_remote_connect + 488
9   org.fritzing.Fritzing         	0x000000010c9f20f7 PartsChecker::newPartsAvailable(QString const&, QString const&, bool, QString&, PartsCheckerResult&) + 2103
10  org.fritzing.Fritzing         	0x000000010c9e9796 UpdateDialog::releasesAvailableSlot() + 710
11  org.qt-project.QtCore         	0x000000010e4215bd QMetaObject::activate(QObject*, int, int, void**) + 749
12  org.fritzing.Fritzing         	0x000000010c9ef424 VersionChecker::finished(QNetworkReply*) + 244
13  org.qt-project.QtCore         	0x000000010e4215bd QMetaObject::activate(QObject*, int, int, void**) + 749
14  org.qt-project.QtNetwork      	0x000000010e08a90a 0x10e05c000 + 190730
15  org.qt-project.QtCore         	0x000000010e4215bd QMetaObject::activate(QObject*, int, int, void**) + 749
16  org.qt-project.QtNetwork      	0x000000010e0a659d 0x10e05c000 + 304541
17  org.qt-project.QtCore         	0x000000010e41a6dc QObject::event(QEvent*) + 156
18  org.qt-project.QtWidgets      	0x000000010d40ab76 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 294
19  org.qt-project.QtWidgets      	0x000000010d40d4c0 QApplication::notify(QObject*, QEvent*) + 5648
20  org.fritzing.Fritzing         	0x000000010c8307fd FApplication::notify(QObject*, QEvent*) + 29
21  org.qt-project.QtCore         	0x000000010e3ee3f4 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 164
22  org.qt-project.QtCore         	0x000000010e3ef014 QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 596
23  libqcocoa.dylib               	0x000000010f9c0bae 0x10f99e000 + 142254
24  libqcocoa.dylib               	0x000000010f9c1481 0x10f99e000 + 144513
25  com.apple.CoreFoundation      	0x00007fff960e0241 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
26  com.apple.CoreFoundation      	0x00007fff960dfaad __CFRunLoopDoSources0 + 253
27  com.apple.CoreFoundation      	0x00007fff961068d9 __CFRunLoopRun + 905
28  com.apple.CoreFoundation      	0x00007fff96106216 CFRunLoopRunSpecific + 230
29  com.apple.HIToolbox           	0x00007fff9404f4ff RunCurrentEventLoopInMode + 277
30  com.apple.HIToolbox           	0x00007fff94056c21 ReceiveNextEventCommon + 355
31  com.apple.HIToolbox           	0x00007fff94056aae BlockUntilNextEventMatchingListInMode + 62
32  com.apple.AppKit              	0x00007fff9250c191 _DPSNextEvent + 659
33  com.apple.AppKit              	0x00007fff9250ba95 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 135
34  com.apple.AppKit              	0x00007fff928e148e NSCoreDragBlockingProc + 103
35  com.apple.HIServices          	0x00007fff973878fd SampleMouseAndKeyboard + 103
[... 510 lines cut ...]
__DATA                             14.7M
__IMAGE                            1256K
__LINKEDIT                         51.7M
__TEXT                            125.3M
__UNICODE                           544K
mapped file                        36.9M
shared memory                       312K
===========                      =======
TOTAL                             572.1M
TOTAL, minus reserved VM space    572.1M

Model: iMac6,1, BootROM IM61.0093.B07, 2 processors, Intel Core 2 Duo, 2.16 GHz, 4 GB, SMC 1.10f3
Graphics: NVIDIA GeForce 7300 GT, NVIDIA GeForce 7300 GT, PCIe, 128 MB
Memory Module: BANK 0/DIMM0, 2 GB, DDR2 SDRAM, 667 MHz, 0x7F4F000000000000, 0x4A4D3636375153552D324720202020202020
Memory Module: BANK 1/DIMM1, 2 GB, DDR2 SDRAM, 667 MHz, 0x7F4F000000000000, 0x4A4D3636375153552D324720202020202020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x87), Broadcom BCM43xx 1.0 (5.10.131.36.11)
Bluetooth: Version 2.5.0f17, 2 service, 19 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: ST3160023AS, 160.04 GB
Parallel ATA Device: MATSHITADVD-R   UJ-85J
USB Device: SABRENT, 0x152d  (JMicron Technology Corp.), 0x1561, 0xfd300000 / 2
USB Device: USB Receiver, 0x046d  (Logitech Inc.), 0xc534, 0x1d100000 / 2
USB Device: Bluetooth USB Host Controller, apple_vendor_id, 0x8206, 0x7d100000 / 2
USB Device: IR Receiver, apple_vendor_id, 0x8240, 0x7d200000 / 3
```

Christophe59, 2018-04-02:
I have the same issue on macOS 10.7.5.
Was there any fix published for it?
Thanks

ejonesss, 2019-06-05:
i have the same problem i found i could keep the program working if i run it from the disk image.

i suspect it is a bug in the writing to the application or the execution of the check for updates upon startup.

to run fritzing from the image you have to copy the application to the desktop then make alias (as you cant make alias on read only disks)

then delete the the copy of the application on the desktop and empty trash

then open fritzing from the alias and using the fix alias point it to the application on the read only image.

you can then unmount the image and the next time you open fritzing it will mount the image each time.

if you skipped the verify upon initial mount it will do it again.

let it fully verify then  future verifies will be much faster.

KjellMorgenstern, 2019-07-18:
Can you reproduce the problem with the latest development build from github releases?

KjellMorgenstern, 2020-01-04:
I assume this has been fixed by 0.9.3->0.9.4 improvements on finding parts and creating parts.db

closed 2020-01-04 by KjellMorgenstern