fritzing/fritzing-app #3327: Should not replace symlink with dir

thecarp, 2017-04-11:
I am running in a vagrant environment (ubuntu/trusty64) where persistent files are presented via a path outside of the home directory. 

So I moved the contents of /home/vagrant/Documents/Fritzing to /vagrant/ and replaced the ~/Documents/Fritzing directory with a symbolic link. ( cd Documents; rmdir Fritzing; ln -s /vagrant Fritzing )

When I ran fritzing, it removed my symbolic link, and recreated the directory, back on the non-persistent storage.

el-j, 2017-04-12:
ok so you install fritzing in vagrant root.
and when you start the application, it creats a fritzing directory in the ~/Documents/Fritzing ?

if this is the problem, why you need the files from the fritzing-"temp" folder?
corret me if i am wrong, this folder is just for temp-files during you run fritzing. a simple workaround could be to copy the files from the folder shortly before you shutdown the vagrant machine.

on restart you can easily put them back where they belong.
this could be done with a single line in a startup-script.

thecarp, 2017-04-12:
Sorry, I should have been more clear. I don't need support for myself. The correct long-term solution is to properly define the directory as a synced dir so the files get used from the host transparently. I was only trying a symlink as a first cut at fixing it before I edited the Vagrant config. I am, personally, all set. 

However, I must argue this is a bug, as symlinks are almost always a direct reflection of intentional action by a user, and usually one who understands what he is doing. There are very few occasions where treating a symlink as a special case makes sense.

Also, just in general, while robustness is a noble goal, I am not sure that blindly removing files from a user's Documents directory is the way to achieve it.

KjellMorgenstern, 2019-05-03:
Would this approach solve the situation: https://github.com/fritzing/fritzing-app/issues/3454 ?

failiz, 2021-11-02:
Is this issue still relevant?

KjellMorgenstern, 2021-11-02:
Fritzing should either complain, or follow the symlink. I don't think replacing it without further notice is a good idea.