fritzing/fritzing-app #3329: crash when both bus and subpart in the same part

mMerlin, 2017-04-13:
I have been exploring Fritzing part creation, and found a problem.  If both internal connections and split schematic are implemented in the same part, Fritzing crashes after placing the part.  It crashes faster if one of the subparts is moved.

In the sample (crashing) part, hiding (block comment) the buses element in the fzp file gets the subparts to work.  Hiding the schematic-subparts element and unhiding the schematic element in the svg file gets the internal connections to work.

Environment:
Fedora 24
x86_64
Fritzing Version 0.9.2 (b8d2d5970658f0bed09c661c9

I have not upgraded to 0.9.3, since trying to stay with what is provided in the Fedora repos, and the new version is not even in the testing repo yet.

fzpz part files rename to zip, to get upload to work.
[SPST-D31A-Relay-crash.zip](https://github.com/fritzing/fritzing-app/files/920929/SPST-D31A-Relay-crash.zip)
[SPST-D31A-Relay-split.zip](https://github.com/fritzing/fritzing-app/files/920956/SPST-D31A-Relay-split.zip)
[SPST-D31A-Relay-bus.zip](https://github.com/fritzing/fritzing-app/files/920957/SPST-D31A-Relay-bus.zip)

jgartrel, 2019-05-03:
I have stumbled across this issue as well.  
Fritzing Version 0.9.3 (b5c895d327c44a3114e5fcc9d8260daf0cbb52806 2016-04
MacOSX

KjellMorgenstern, 2019-05-04:
@jgartrel Did you try with same files as mMerlin? Or could you upload your files and reproduction steps as well? We can not look into 0.9.2 issues (number of open issues is just to big for that), but if the crash can reproduced on 0.9.3 this is a different thing.

jgartrel, 2019-05-08:
@KjellMorgenstern , I used a different part that I had created.  I will attempt to attach it here.  Please forgive the incorrect author attribution as the files were copied from the Sparkfun parts and are not completely done yet.  To reproduce the error change the <nobuses></nobuses> tags to <buses></buses> in the part file.

fzpz part files rename to zip, to get upload to work.
[DMC6040SSD.zip](https://github.com/fritzing/fritzing-app/files/3155776/DMC6040SSD.zip)

jgartrel, 2019-05-08:
The part is a N and P channel mosfet in a single package: https://www.diodes.com/assets/Datasheets/DMC6040SSD.pdf

If this issue gets resolved, I would be more than happy to clean up this part (and the other dual N channel and dual P-channel parts that I have created), and add them, as well as generic counterparts, to the Fritzing library.

KjellMorgenstern, 2019-05-08:
I tried the part on the current development version. It loads fine, and I was able to assemble a small circuit, adding some capacitors. I moved arround parts on the breadboard and the schematics. Note that in the schematics, the DMC6040 is currently just two parts (one for each MOSFET). 
I got it crashing most of the time when using the autorouter from the schematics view. This might be a different crash than the one you are exeriencing, since you did not write anything about using the autorouter?

The backtrace is
```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007f2b59ec3535 in __GI_abort () at abort.c:79
#2  0x00007f2b59f2a726 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7f2b5a050952 "%s\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007f2b59f3159a in malloc_printerr (str=str@entry=0x7f2b5a04e9b0 "corrupted size vs. prev_size") at malloc.c:5352
#4  0x00007f2b59f31841 in unlink_chunk (p=<optimized out>, av=0x7f2b5a082c40 <main_arena>) at malloc.c:1464
#5  0x00007f2b59f32a18 in malloc_consolidate (av=av@entry=0x7f2b5a082c40 <main_arena>) at malloc.c:4510
#6  0x00007f2b59f33740 in _int_free (av=0x7f2b5a082c40 <main_arena>, p=0x56116c39c400, have_lock=<optimized out>) at malloc.c:4408
#7  0x000056115c0ed7da in __gnu_cxx::new_allocator<GridPoint>::deallocate (this=0x7ffe4ede7d48, __p=0x56116c39c410) at /usr/include/c++/5/ext/new_allocator.h:110
#8  0x000056115c0eaa7a in __gnu_cxx::__alloc_traits<std::allocator<GridPoint> >::deallocate (__a=..., __p=0x56116c39c410, __n=2048) at /usr/include/c++/5/ext/alloc_traits.h:185
#9  0x000056115c0e603e in std::_Vector_base<GridPoint, std::allocator<GridPoint> >::_M_deallocate (this=0x7ffe4ede7d48, __p=0x56116c39c410, __n=2048) at /usr/include/c++/5/bits/stl_vector.h:178
#10 0x000056115c0e5621 in std::_Vector_base<GridPoint, std::allocator<GridPoint> >::~_Vector_base (this=0x7ffe4ede7d48, __in_chrg=<optimized out>) at /usr/include/c++/5/bits/stl_vector.h:160
#11 0x000056115c0dfeb8 in std::vector<GridPoint, std::allocator<GridPoint> >::~vector (this=0x7ffe4ede7d48, __in_chrg=<optimized out>) at /usr/include/c++/5/bits/stl_vector.h:425
#12 0x000056115c0de14c in std::priority_queue<GridPoint, std::vector<GridPoint, std::allocator<GridPoint> >, std::less<GridPoint> >::~priority_queue (this=0x7ffe4ede7d48, __in_chrg=<optimized out>)
    at /usr/include/c++/5/bits/stl_queue.h:370
#13 0x000056115c0de50d in RouteThing::~RouteThing (this=0x7ffe4ede7cc0, __in_chrg=<optimized out>) at src/autoroute/mazerouter/mazerouter.h:144
#14 0x000056115c0cc8c9 in MazeRouter::routeNets (this=0x56116594eb20, netList=..., makeJumper=false, currentScore=..., gridSize=..., allOrderings=...) at src/autoroute/mazerouter/mazerouter.cpp:1002
#15 0x000056115c0c979d in MazeRouter::start (this=0x56116594eb20) at src/autoroute/mazerouter/mazerouter.cpp:788
#16 0x000056115be26a3d in MainWindow::newAutoroute (this=0x561164a7e0c0) at src/mainwindow/mainwindow_menu.cpp:2985
#17 0x000056115c224b52 in MainWindow::qt_static_metacall (_o=0x561164a7e0c0, _c=QMetaObject::InvokeMetaMethod, _id=120, _a=0x7ffe4ede8490) at debug/moc_mainwindow.cpp:1288
#18 0x00007f2b5a691426 in QMetaObject::activate(QObject*, int, int, void**) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007f2b5b280fe2 in QAction::triggered(bool) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#20 0x00007f2b5b2835f0 in QAction::activate(QAction::ActionEvent) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007f2b5b3715ad in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#22 0x00007f2b5b3717e5 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#23 0x00007f2b5b45b6ca in QToolButton::mouseReleaseEvent(QMouseEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x000056115bdbe9df in SketchToolButton::mouseReleaseEvent (this=0x561160827b40, event=0x7ffe4ede8af0) at src/sketchtoolbutton.cpp:122
#25 0x00007f2b5b2c6b58 in QWidget::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#26 0x00007f2b5b45b773 in QToolButton::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#27 0x00007f2b5b287551 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#28 0x00007f2b5b28eb77 in QApplication::notify(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#29 0x000056115bda3bda in FApplication::notify (this=0x56115d5b91d0, receiver=0x561160827b40, e=0x7ffe4ede8af0) at src/fapplication.cpp:1611
#30 0x00007f2b5a6678e9 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#31 0x00007f2b5b28debf in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#32 0x00007f2b5b2e157b in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#33 0x00007f2b5b2e453f in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#34 0x00007f2b5b287551 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#35 0x00007f2b5b28e930 in QApplication::notify(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#36 0x000056115bda3bda in FApplication::notify (this=0x56115d5b91d0, receiver=0x56115ffb0250, e=0x7ffe4ede9030) at src/fapplication.cpp:1611
[... cut ...]
#44 0x00007f2b5969cddc in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#45 0x00007f2b5a6bb047 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#46 0x00007f2b5a6665bb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#47 0x00007f2b5a66e5e2 in QCoreApplication::exec() () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#48 0x000056115bdbd6db in main (argc=1, argv=0x7ffe4ede95a8) at src/main.cpp:148

```

jgartrel, 2019-05-08:
@KjellMorgenstern , Looking back at my posting 9 hours ago.  It looks like github removed the buses and nobuses tag references that I made (It probably thought they were improperly formatted HTML).  I am guessing that because of that, you were using the part without having the internal connections enabled.  If you make the changes described in the diff above, I am guessing you will see the part crash Fritzing.

jgartrel, 2019-05-08:
LoL,  I guess I meant to say the diff below, since somehow I posted it previously in the future:

"jgartrel commented 5 hours from now:  So, on my mac using the version above ..."

KjellMorgenstern, 2019-05-08:
With thelast part modification to add buses instead nobuses, I can reproduce the crash when adding the part to the breadboard.

KjellMorgenstern, 2019-05-08:
With the latest modification to the part (adding the buses), I  can reproduce the crash.

jgartrel, 2019-05-08:
Fantastic, Let me know if you need anything else.  and Thank you!

KjellMorgenstern, 2019-05-08:
Seems like github had a hickup.

KjellMorgenstern, 2019-06-29:
A hopefully more useful backtrace (crash after the part was put on the breadboard)

<pre>(gdb) 
#0  <font color="#3465A4">0x0000563eece70880</font> in <font color="#C4A000">QHash&lt;int, QPointer&lt;ConnectorItem&gt; &gt;::size</font> (<font color="#06989A">this</font>=0x2590000021b) at <font color="#4E9A06">/usr/include/x86_64-linux-gnu/qt5/QtCore/qhash.h</font>:267
#1  <font color="#3465A4">0x0000563eece6fd64</font> in <font color="#C4A000">QHash&lt;int, QPointer&lt;ConnectorItem&gt; &gt;::values</font> (<font color="#06989A">this</font>=0x2590000021b) at <font color="#4E9A06">/usr/include/x86_64-linux-gnu/qt5/QtCore/qhash.h</font>:711
#2  <font color="#3465A4">0x0000563eece6f45e</font> in <font color="#C4A000">Connector::viewItems</font> (<font color="#06989A">this</font>=0x259000001fb) at <font color="#4E9A06">src/connectors/connector.cpp</font>:277
#3  <font color="#3465A4">0x0000563eecd49159</font> in <font color="#C4A000">ItemBase::busConnectorItems</font> (<font color="#06989A">this</font>=0x563efadbffa0, <font color="#06989A">bus</font>=0x563efa86f420, <font color="#06989A">fromConnectorItem</font>=0x563efa4f6c60, <font color="#06989A">items</font>=...) at <font color="#4E9A06">src/items/itembase.cpp</font>:761
#4  <font color="#3465A4">0x0000563eece77e25</font> in <font color="#C4A000">ConnectorItem::collectEqualPotential</font> (<font color="#06989A">connectorItems</font>=..., <font color="#06989A">crossLayers</font>=true, <font color="#06989A">skipFlags</font>=...) at <font color="#4E9A06">src/connectors/connectoritem.cpp</font>:1356
#5  <font color="#3465A4">0x0000563eece7373f</font> in <font color="#C4A000">ConnectorItem::restoreColor</font> (<font color="#06989A">this</font>=0x563efa4f6c60, <font color="#06989A">visited</font>=...) at <font color="#4E9A06">src/connectors/connectoritem.cpp</font>:537
#6  <font color="#3465A4">0x0000563eece72f5d</font> in <font color="#C4A000">ConnectorItem::connectTo</font> (<font color="#06989A">this</font>=0x563efa4f6c60, <font color="#06989A">connected</font>=0x563efa2b45e0) at <font color="#4E9A06">src/connectors/connectoritem.cpp</font>:472
#7  <font color="#3465A4">0x0000563eecf016bc</font> in <font color="#C4A000">SketchWidget::changeConnectionAux</font> (<font color="#06989A">this</font>=0x563ef0343430, <font color="#06989A">fromID</font>=854205370, <font color="#06989A">fromConnectorID</font>=..., <font color="#06989A">toID</font>=57680, <font color="#06989A">toConnectorID</font>=..., <font color="#06989A">viewLayerPlacement</font>=ViewLayer::NewBottom, <font color="#06989A">connect</font>=true, 
    <font color="#06989A">updateConnections</font>=true) at <font color="#4E9A06">src/sketch/sketchwidget.cpp</font>:5064
#8  <font color="#3465A4">0x0000563eecf00f15</font> in <font color="#C4A000">SketchWidget::changeConnection</font> (<font color="#06989A">this</font>=0x563ef0343430, <font color="#06989A">fromID</font>=854205370, <font color="#06989A">fromConnectorID</font>=..., <font color="#06989A">toID</font>=57680, <font color="#06989A">toConnectorID</font>=..., <font color="#06989A">viewLayerPlacement</font>=ViewLayer::NewBottom, <font color="#06989A">connect</font>=true, 
    <font color="#06989A">doEmit</font>=true, <font color="#06989A">updateConnections</font>=true) at <font color="#4E9A06">src/sketch/sketchwidget.cpp</font>:5006
#9  <font color="#3465A4">0x0000563eecb04f16</font> in <font color="#C4A000">ChangeConnectionCommand::redo</font> (<font color="#06989A">this</font>=0x563f0173f850) at <font color="#4E9A06">src/commands.cpp</font>:505
#10 <font color="#3465A4">0x00007fd701802cc2</font> in <font color="#C4A000">QUndoCommand::redo()</font> () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 <font color="#3465A4">0x0000563eecb12eea</font> in <font color="#C4A000">TemporaryCommand::redo</font> (<font color="#06989A">this</font>=0x563f0170ecb0) at <font color="#4E9A06">src/commands.cpp</font>:2253
#12 <font color="#3465A4">0x00007fd701804e93</font> in <font color="#C4A000">QUndoStack::push(QUndoCommand*)</font> () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 <font color="#3465A4">0x0000563eecb50514</font> in <font color="#C4A000">WaitPushUndoStack::push</font> (<font color="#06989A">this</font>=0x563ef1a17be0, <font color="#06989A">cmd</font>=0x563f0170ecb0) at <font color="#4E9A06">src/waitpushundostack.cpp</font>:83
#14 <font color="#3465A4">0x0000563eecb50254</font> in <font color="#C4A000">CommandTimer::timedout</font> (<font color="#06989A">this</font>=0x563f0170fa50) at <font color="#4E9A06">src/waitpushundostack.cpp</font>:48
#15 <font color="#3465A4">0x0000563eecf7328c</font> in <font color="#C4A000">CommandTimer::qt_static_metacall</font> (<font color="#06989A">_o</font>=0x563f0170fa50, <font color="#06989A">_c</font>=QMetaObject::InvokeMetaMethod, <font color="#06989A">_id</font>=0, <font color="#06989A">_a</font>=0x7ffc11847940) at <font color="#4E9A06">debug/moc_waitpushundostack.cpp</font>:141
#16 <font color="#3465A4">0x00007fd7008a6426</font> in <font color="#C4A000">QMetaObject::activate(QObject*, int, int, void**)</font> () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#17 <font color="#3465A4">0x00007fd7008b20c7</font> in <font color="#C4A000">QTimer::timeout(QTimer::QPrivateSignal)</font> () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#18 <font color="#3465A4">0x00007fd7008a6c5b</font> in <font color="#C4A000">QObject::event(QEvent*)</font> () from /lib/x86_64-linux-gnu/libQt5Core.so.5
[... cut ...]
#29 <font color="#3465A4">0x00007fd70087b5bb</font> in <font color="#C4A000">QEventLoop::exec(QFlags&lt;QEventLoop::ProcessEventsFlag&gt;)</font> () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#30 <font color="#3465A4">0x00007fd7008835e2</font> in <font color="#C4A000">QCoreApplication::exec()</font> () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#31 <font color="#3465A4">0x0000563eecb4387d</font> in <font color="#C4A000">main</font> (<font color="#06989A">argc</font>=1, <font color="#06989A">argv</font>=0x7ffc11848148) at <font color="#4E9A06">src/main.cpp</font>:132
</pre>

KjellMorgenstern, 2020-11-11:
Retested on dev version with this part, mo crash. There were some uninitialized values and other fixes, but it is not 100% clear what caused the issue. Should be retested with 

[DMC6040SSD_7818bd58bd3d04e38121346b52ef4a24_1.fzp.zip](https://github.com/fritzing/fritzing-app/files/5523227/DMC6040SSD_7818bd58bd3d04e38121346b52ef4a24_1.fzp.zip)

mMerlin, 2020-11-14:
With the notification about the latest comment here, I revisited the original case. The previous relay part still crashes with 0.9.4, but I have learned more about parts creation since that was created. A rebuild of the part works (so far) with 0.9.4. Most of the cleanup was in the svg files. I resequenced the connector numbers in the fzp, and adjusted a lot of text description content, but the only 'functional' differences I see are, 2 sub-parts instead of 3, and the duplicate bused parts have been marked as hybrid (and hidden on the schematic view). I do not see what is triggering the crash in the earlier version of the part, but this one is working. I still have some more cleanup (and testing) to do on it, but here is a working copy for reference/comparison.

[SPST-D31A-Relay_1.fzpz.zip](https://github.com/fritzing/fritzing-app/files/5540041/SPST-D31A-Relay_1.fzpz.zip)

failiz, 2021-03-18:
already released, close

failiz, 2021-03-18:
I wrote a message with "already released, close" and I deleted it when I saw the last comment. Maybe we should test this in 0.9.6 before closing it.

KjellMorgenstern, 2021-03-18:
Removing the solved lable until this is retested

KjellMorgenstern, 2024-03-20:
Retested with 0.9.9 and 1.0.2

closed 2024-03-20 by KjellMorgenstern