fritzing/fritzing-app #3330: Generic resistor with 100mil footprint (non obstructing top view) also in breadboard view

albert8855, 2017-04-15:
Dear fritzing-Team,

fritzing is in principle a quite suitable tool for designing and documenting circuits on stripboard/perfboard. I like the photo realistic appearence which is nice for compiling instructables.

However, what I am missing desperately is **a generic resistor which has a correct upright standing footprint also in the stripboard/perfboard (breadboard) view**. Though it is possible to choose for example the 100mil spacing in the inspector, this shorter spacing is only reflected in the PCB view while on the stripboard the 400mil image persists. 

Tilting the 400 mil image and bending of the leads down is not an option because the resistor is overlapping too much other parts on the stripboard (I tried that and got a mess). I know that I can create my own custom parts (one for each resistance value) and I did that already but my request here is about a **generic** part. 

What could be done to solve that problem? The color code may be dropped.

Best Regards
Albert

flaix, 2020-10-18:
Such a part can be found in the Fritzing forums, made by the community.
Another approach would be to change the existing core resistor to change the image in breadboard view, too, when set to 100 mil standing in the inspector.

One open question for the latter would be, what is the expected behaviour of a placed resistor on the breadboard when switching from upright to horizontal position. Imagine a few resistors have been placed and connected on the breadboard, an then in PCB view the orientation is change to upright. What happens to the resistors in breadboard view? Should Fritzing try to keep the connections on the breadboard? If so, how, with wires or using bendable legs? Should only the image be changed, risking to disconnect them all on the breadboard?

failiz, 2021-01-08:
There are several bus that report this issue (pitch changes in the PCB view but not in the breadboard view). I also found it strange. 
For resistors, capacitors and similar, I think that the best would be to have two properties: the pitch in the breadboard and the pitch in the PCB view. So, you can change just one property without changing the other.
For connectors and pin headers, I think it would be better to have only one property and change both views.

failiz, 2021-01-08:
Same issue as #2923 
I mentioned it here as I found strange it: #2576
It is also related to how fritzing handles the properties: one part has one view in BB, sch and PCB. If you want to change one view you have to change another part. Related to #3775

LisaChyhryna-fritzing, 2023-08-28:
[image]
[image]
[image]