fritzing/fritzing-app #3333: NEW GerberOutput seems broken

el-j, 2017-04-25:
There is a problem with the new gerber output when the pcb has copper or ground-filling:

screenshot from gerbV:
[image]

it seems that it is a problem with circles, they are not closed properly:

[image]

@nraynaud @Kazade do you have any idea why this happens?

if i look at the gerberout in a text editor i cannot see any circle commands G02/G03 or G75,
shouldn't there be just 2 circles in the last screenshot?

thanks for your help! 

```
%ASAXBY*%
%FSLAX26Y26*%
%MOIN*%
%OFA0B0*%
%SFA1.0B1.0*%
G01*
G36*
X564200Y442349D02*
X568100Y441149D01*
X574200Y437149D01*
X578400Y431049D01*
X579600Y426949D01*
X579900Y423049D01*
X579600Y419149D01*
X578400Y415149D01*
X574200Y409149D01*
X568200Y404949D01*
X564200Y403749D01*
X560300Y403349D01*
X556300Y403749D01*
X552300Y404949D01*
X546200Y409149D01*
X542200Y415149D01*
X541000Y419049D01*
X540600Y423049D01*
X541000Y427049D01*
X542200Y430949D01*
X546200Y437149D01*
X552400Y441149D01*
X556300Y442349D01*
X560300Y442749D01*
%LPC*%
G36*
X558300Y430649D02*
X557400Y430449D01*
X554700Y428649D01*
X552900Y425949D01*
X552700Y425249D01*
X552500Y423049D01*
X552700Y421249D01*
X553000Y420249D01*
X554700Y417649D01*
X557600Y415649D01*
X558300Y415449D01*
X560200Y415249D01*
X562300Y415449D01*
X562900Y415649D01*
X565700Y417649D01*
X567800Y420649D01*
X567900Y421249D01*
X568100Y423049D01*
X567900Y425249D01*
X567700Y425749D01*
X565700Y428649D01*
X563100Y430449D01*
X562300Y430649D01*
X560200Y430849D01*
%LPD*%
G36*
X201600Y459449D02*
X205500Y458249D01*
X211700Y454249D01*
X215600Y448149D01*
X217000Y444149D01*
X217300Y440149D01*
X217000Y436149D01*
X215600Y432149D01*
X211700Y426249D01*
X205600Y422049D01*
X201600Y420849D01*
X197700Y420449D01*
X193700Y420849D01*
X189700Y422049D01*
X183600Y426249D01*
X179600Y432249D01*
X178400Y436149D01*
X178000Y440149D01*
X178400Y444149D01*
X179600Y448049D01*
X183600Y454249D01*
X189800Y458249D01*
X193700Y459449D01*
X197700Y459849D01*
%LPC*%
G36*
X195700Y447749D02*
X194800Y447549D01*
X192100Y445749D01*
X190300Y443049D01*
X190100Y442349D01*
X189900Y440149D01*
X190100Y438349D01*
X190400Y437349D01*
X192100Y434749D01*
X195000Y432749D01*
X195700Y432549D01*
X197600Y432349D01*
X199700Y432549D01*
X200300Y432749D01*
X203200Y434749D01*
X204900Y437349D01*
X205400Y438749D01*
X205500Y440149D01*
X205400Y441749D01*
X204900Y443149D01*
X203200Y445749D01*
X200500Y447549D01*
X199700Y447749D01*
X197600Y447949D01*
%LPD*%
G37*
M02*
```

nraynaud, 2017-04-25:
clipper polygonizes the circles.

nraynaud, 2017-04-25:
I'm sorry, but I don't even have a platform to develop for fritzing, I would need days to get a new working copy up and running. it looks like all the polygons are chained without "lifting the pen".

AHorneffer, 2017-12-03:
The gerber file from the original post looks fine when I view it with gerbview (from KiCad), but not when I view it with Gerbv. 
From: https://www.ucamco.com/files/downloads/file/81/the_gerber_file_format_specification.pdf it looks to me like having a "G36*" command followed followed by another "G36*" command without a "G37*" command is not correct.  (Aka it violates the syntax of gerber files.) 
When I add the missing "G37*" commands, then it looks fine in both programs.

nraynaud, 2017-12-03:
@AHorneffer thank you for your work

Intrinsically-Sublime, 2017-12-04:
I can confirm @AHorneffer  patch does indeed fix this issue.

aknoerig, 2017-12-04:
Fantastic, thanks a lot! It's now merged.

closed 2017-12-04 by aknoerig