fritzing/fritzing-app #3334: Breadboard, Schematic and PCB views hangs-up (not responding)        (Version 0.9.3)

faceter, 2017-05-01:
When any activity is started in these views Ffitzing hangs up for a time eventually clearing until the next attempt at use takes place.
This is an upated installation  on a PC running Windows 10.

This makes it impossible for me (a newby to Fritzing) to progress with a project.   Your help would be appreciated.

georgielaa, 2017-05-18:
hi i have just installed on windows 10 and i get the same issue. If you run the fritzing application with the compatibility troubleshooter and use windows 8 settings, that seems to fix the hanging issue. hope this helps you out :)

Gerriko, 2017-05-28:
Maybe this is the issue. I had a similar problem on my Windows 10 PC but waited (very) patiently. Eventually responded to tell me there was a fritzing library update available. Maybe some part of the check for updates routine is blocking.

KjellMorgenstern, 2019-05-02:
Are you able to run Fritzing on a machine that is disconnected from the internet?

KjellMorgenstern, 2019-07-09:
You could try the development version of Fritzing. Please don't use for production (backup your Fritzing project before download), as it is not yet widely testet. https://github.com/fritzing/fritzing-app/releases/tag/CD-288
As the new build is done on Windows 10, and it seems no other platforms were affected, I am closing this issue. Please reopen if you still observe the issue with latest builds.

closed 2019-07-09 by KjellMorgenstern