fritzing/fritzing-app #3336: White text over light gray background on menus

pabloab, 2017-05-08:
## Ubuntu 16.04.2 LTS, Fritzing v0.9.2 from repos:
[image]

All menus of the UI on Fritzing, the row onmouseover event change font color to white, which makes very difficult to read the selected value with the default light gray background.

KY64, 2017-09-13:
Not just there, it's also happen on all "Code Window" submenu
[image]

Oozenthor, 2019-06-22:
Happens in the Inspector/Properties drop downs as well. Looks like Fritzing does not play nice with Linux dark themes (Titlebar / backgrounds are gray instead of black like my theme is asking for).

KjellMorgenstern, 2020-02-06:
Could not reproduce on 0.9.5d. Either this is already fixed, or I didn't find the right dark theme for this. If the issue persists, please reopen and provide desktop settings (Windowmanager, Theme...)

closed 2020-02-06 by KjellMorgenstern