fritzing/fritzing-app #3342: Segfault on 64-bit Devuan linux

np511, 2017-06-26:
I installed Fritzing on Devuan (Debian w/o systemd) and launched it. It crashed.

I used GDB, installed some debug libraries and got the following:

 gdb fritzing 
GNU gdb (Debian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from fritzing...(no debugging symbols found)...done.
(gdb) r
Starting program: /usr/bin/fritzing 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe8815700 (LWP 14464)]
[New Thread 0x7fffdee54700 (LWP 14467)]
[New Thread 0x7fffde653700 (LWP 14468)]
[New Thread 0x7fffd7fff700 (LWP 14469)]

Thread 1 "fritzing" received signal SIGSEGV, Segmentation fault.
0x00007ffff3ee4215 in findConverter (pErrorCode=0x7fffff7ff0dc, containsOption=0x7fffff7ff0db "\001", alias=0x7fffff7ff250 "US-ASCII") at ucnv_io.cpp:557
557	ucnv_io.cpp: No such file or directory.
(gdb) bt
#0  0x00007ffff3ee4215 in findConverter (pErrorCode=0x7fffff7ff0dc, containsOption=0x7fffff7ff0db "\001", alias=0x7fffff7ff250 "US-ASCII") at ucnv_io.cpp:557
#1  ucnv_io_getConverterName_57 (alias=<optimized out>, containsOption=containsOption@entry=0x7fffff7ff0db "\001", pErrorCode=pErrorCode@entry=0x7fffff7ff0dc) at ucnv_io.cpp:754
#2  0x00007ffff3ee236b in ucnv_loadSharedData_57 (converterName=<optimized out>, pPieces=pPieces@entry=0x7fffff7ff250, pArgs=pArgs@entry=0x7fffff7ff220, err=err@entry=0x7fffff7ff37c)
    at ucnv_bld.cpp:777
#3  0x00007ffff3ee2620 in ucnv_createConverter_57 (myUConverter=0x0, converterName=<optimized out>, err=0x7fffff7ff37c) at ucnv_bld.cpp:832
#4  0x00007ffff69e1cba in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007ffff69e2594 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007ffff683eb8f in QString::fromLocal8Bit_helper(char const*, int) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#7  0x00007ffff690564c in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007ffff6906ba2 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff6907256 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007ffff68a12ec in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x00007ffff68a1a95 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007ffff68a234c in QDirIterator::QDirIterator(QString const&, QStringList const&, QFlags<QDir::Filter>, QFlags<QDirIterator::IteratorFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff689b34c in QDir::entryList(QStringList const&, QFlags<QDir::Filter>, QFlags<QDir::SortFlag>) const () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff689b9cf in QDir::entryList(QFlags<QDir::Filter>, QFlags<QDir::SortFlag>) const () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x0000555555930741 in ?? ()
#16 0x00005555559307e6 in ?? ()
#17 0x00005555559307e6 in ?? ()
#18 0x00005555559307e6 in ?? ()
#19 0x00005555559307e6 in ?? ()
#20 0x00005555559307e6 in ?? ()
#21 0x00005555559307e6 in ?? ()
#22 0x00005555559307e6 in ?? ()
#23 0x00005555559307e6 in ?? ()
#24 0x00005555559307e6 in ?? ()
#25 0x00005555559307e6 in ?? ()
#26 0x00005555559307e6 in ?? ()
#27 0x00005555559307e6 in ?? ()
#28 0x00005555559307e6 in ?? ()
#29 0x00005555559307e6 in ?? ()
#30 0x00005555559307e6 in ?? ()
#31 0x00005555559307e6 in ?? ()
#32 0x00005555559307e6 in ?? ()
#33 0x00005555559307e6 in ?? ()
#34 0x00005555559307e6 in ?? ()
#35 0x00005555559307e6 in ?? ()
#36 0x00005555559307e6 in ?? ()
#37 0x00005555559307e6 in ?? ()
#38 0x00005555559307e6 in ?? ()
#39 0x00005555559307e6 in ?? ()
#40 0x00005555559307e6 in ?? ()
#41 0x00005555559307e6 in ?? ()
#42 0x00005555559307e6 in ?? ()
#43 0x00005555559307e6 in ?? ()
#44 0x00005555559307e6 in ?? ()
#45 0x00005555559307e6 in ?? ()

alexws54tk, 2017-06-27:
I compiled Fritzing on Gentoo and launched it.
```
% Fritzing --help

Fritzing version  "0.9.3b.04.19.5c895d327c44a3114e5fcc9d8260daf0cbb52806"  , Qt version  5.7.1 
```
It crashed.
```
% Fritzing
QIODevice::read (QFile, "/home/alex/bins/core.fzb"): device not open
QIODevice::read (QFile, "/home/alex/bins/contribParts.fzb"): device not open
QIODevice::read (QFile, "/home/alex/bins/core.fzb"): device not open
QIODevice::read (QFile, "/home/alex/bins/contribParts.fzb"): device not open
[1]    17421 segmentation fault  Fritzing
```
I used GDB, installed some debug libraries and got the following:
```
 % gdb Fritzing -d -f /usr/share/fritzing/parts
GNU gdb (Gentoo 8.0 vanilla) 8.0
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://bugs.gentoo.org/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Warning: /usr/share/fritzing/parts/-f: Нет такого файла или каталога.
Reading symbols from Fritzing...(no debugging symbols found)...done.
"/usr/share/fritzing/parts" is not a core dump: Это каталог
(gdb) r
Starting program: /usr/bin/Fritzing 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7fffe9bf7700 (LWP 19324)]
[New Thread 0x7fffdd951700 (LWP 19325)]
[New Thread 0x7fffd56cd700 (LWP 19333)]
[New Thread 0x7fffd4e9c700 (LWP 19334)]
[New Thread 0x7fffcffff700 (LWP 19335)]
[New Thread 0x7fffcf7fe700 (LWP 19338)]
[New Thread 0x7fffceffd700 (LWP 19339)]
[New Thread 0x7fffce7fc700 (LWP 19340)]
libpng warning: iCCP: known incorrect sRGB profile
[Thread 0x7fffd4e9c700 (LWP 19334) exited]
[New Thread 0x7fffd4e9c700 (LWP 19341)]
[New Thread 0x7fffcdffb700 (LWP 19342)]
[New Thread 0x7fffcd7fa700 (LWP 19343)]
[New Thread 0x7fffccff9700 (LWP 19344)]
[Thread 0x7fffceffd700 (LWP 19339) exited]
[New Thread 0x7fffceffd700 (LWP 19345)]
[New Thread 0x7fffabfff700 (LWP 19346)]
[New Thread 0x7fffab7fe700 (LWP 19347)]
[New Thread 0x7fffaaffd700 (LWP 19348)]
[New Thread 0x7fffaa7fc700 (LWP 19349)]
[New Thread 0x7fffa9ffb700 (LWP 19350)]
[Thread 0x7fffcd7fa700 (LWP 19343) exited]
[Thread 0x7fffccff9700 (LWP 19344) exited]
[Thread 0x7fffcdffb700 (LWP 19342) exited]
[Thread 0x7fffd4e9c700 (LWP 19341) exited]
[New Thread 0x7fffd4e9c700 (LWP 19351)]
[New Thread 0x7fffcdffb700 (LWP 19352)]
[New Thread 0x7fffccff9700 (LWP 19353)]
[New Thread 0x7fffcd7fa700 (LWP 19354)]
[New Thread 0x7fffa97fa700 (LWP 19355)]
[Thread 0x7fffcf7fe700 (LWP 19338) exited]
[... 58 lines cut ...]
#0  0x0000000000522f70 in ?? ()
#1  0x000000000051cb37 in ?? ()
#2  0x000000000083fa83 in ?? ()
#3  0x00007ffff6a495b7 in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib64/libQt5Core.so.5
#4  0x00007ffff752c90d in ?? () from /usr/lib64/libQt5Widgets.so.5
#5  0x00007ffff6a495b7 in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib64/libQt5Core.so.5
#6  0x00007fffe8fb597f in ?? () from /usr/lib64/qt5/plugins/platformthemes/KDEPlasmaPlatformTheme.so
#7  0x00007fffe8fc590b in ?? () from /usr/lib64/qt5/plugins/platformthemes/KDEPlasmaPlatformTheme.so
#8  0x00007fffe8fc6108 in ?? () from /usr/lib64/qt5/plugins/platformthemes/KDEPlasmaPlatformTheme.so
#9  0x00007fffea496523 in ?? () from /usr/lib64/libQt5DBus.so.5
#10 0x00007fffea49a80a in ?? () from /usr/lib64/libQt5DBus.so.5
#11 0x00007fffea49b11b in ?? () from /usr/lib64/libQt5DBus.so.5
#12 0x00007fffea49d8de in ?? () from /usr/lib64/libQt5DBus.so.5
#13 0x00007ffff6a4a731 in QObject::event(QEvent*) () from /usr/lib64/libQt5Core.so.5
#14 0x00007ffff73b904c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib64/libQt5Widgets.so.5
#15 0x00007ffff73c09f6 in QApplication::notify(QObject*, QEvent*) () from /usr/lib64/libQt5Widgets.so.5
#16 0x00000000004b6902 in ?? ()
#17 0x00007ffff6a2323a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib64/libQt5Core.so.5
#18 0x00007ffff6a258bb in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /usr/lib64/libQt5Core.so.5
#19 0x00007ffff6a70483 in ?? () from /usr/lib64/libQt5Core.so.5
#20 0x00007ffff304e4e7 in g_main_context_dispatch () from /usr/lib64/libglib-2.0.so.0
#21 0x00007ffff304e740 in ?? () from /usr/lib64/libglib-2.0.so.0
#22 0x00007ffff304e7ec in g_main_context_iteration () from /usr/lib64/libglib-2.0.so.0
#23 0x00007ffff6a704ff in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib64/libQt5Core.so.5
```

np511, 2017-06-28:
When I run it through the menu, it crashes without giving anything past the splash screen. When I ran it in the Downloads folder (I wanted to see if some library upgrades I had done had fixed it before unpacking the tarball I downloaded), I got two windows open:

1: "Cannot read file /home/noam/Downloads/bins/core.fzb:
No such file or directory."
2: Loading new sketch (14%)

Clicking OK on the first one gave a new one: "Fritzing cannot load the parts bin"

However, the app opened (relatively) successfully, with a window warning that 604 parts couldn't be found.

np511, 2017-06-28:
Oops. I forgot to install the fritzing-parts package. I'd assumed it was a dependency of Fritzing, and it isn't. Now that I think about it, I had similar problems before I switched from Debian to Devuan, though I don't quite remember what happened.

np511, 2017-06-28:
Yep. Installing the fritzing-parts package fixed it.

closed 2017-06-28 by np511