fritzing/fritzing-app #3346: Sketches with custom parts included

pintman, 2017-07-30:
Currently a sketch with custom parts is not portable. Parts are referenced with absolute paths and break when transferred to another computer. This makes collaboration difficult.
To address this issue, custom made parts should be included into sketches (maybe optional) and therefore can be used across computers.

Intrinsically-Sublime, 2017-08-17:
Custom parts are saved in the sketch by default. The only reason I could see them not being saved with the sketch would be because you have them in the core parts some how and it thinks that everyone has them. It is very common for people to download other peoples sketches to get custom parts from them. Also for problems like this is would be best to ask on the Fritzing forum as there are more active people there then on Github for Fritzing.

pintman, 2017-08-17:
Sublime schrieb am 17.08.2017 um 04:25:
> The only reason I could
> see them not being saved with the sketch would be because you have them
> in the core parts some how and it thinks that everyone has them. 

I think this was exactly the problem. Thanks for your hint.

closed 2017-08-17 by pintman