fritzing/fritzing-app #3350: Unable to set user directory (Ubuntu & Fritzing 0.9.3)

al3xsh, 2017-09-12:
Hi,

Every time I start Fritzing it creates a $HOME/Fritzing directory and I cannot work out how to create this somewhere else (e.g. in my Dropbox folder).  I've tried searching and have been unable to find anything that works (passing the -f switch to the fritzing executable just ends up giving me loads of errors and then creating the $HOME/Fritzing directory anyway!).

I'm hoping someone can help as, although this isn't a showstopper by any means, it is very annoying! 

Regards,

Alex

Intrinsically-Sublime, 2017-09-15:
You can not change the user directory. You can however replace the Fritzing home directory with a link to your other folder which will mean all the files will end up where you want but you will still have to leave the link in place forever.

sixtyfive, 2017-11-18:
That will not do. My home directory is a sacred place and I do not accept any non-dot-prefix'd files, folders or symlinks in there unless I created them myself. And I can't be the only one who feels like that. For me, this is a showstopper to continue using Fritzing after having it tried out and found otherwise wonderful. Please consider putting the folder in the preferences dialog.

Intrinsically-Sublime, 2017-11-18:
@sixtyfive As you can tell by the commits in the Fritzing app as of late the development has staled. Among the feature requiring attention your request is most likely near the bottom and making demands is not likely going to help. If you feel it is required I would suggest you put in the effort to make the changes yourself. For the people using Fritzing that actually appreciate the software we are grateful for the effort that has been put in so far and can live with the few small issues it has. If development continues the high priority issues will get addressed first like the broken new gerber export in the current Github head. After all the priority issues there will be things like finishing the parts editor which is helpful to everyone using Fritizing that will most likely come before addressing user preferred changes.

KjellMorgenstern, 2019-11-24:
Fritzing follows the standards defined by Qt, which tries to follow standards specified by the OS at use: https://doc.qt.io/qt-5/qstandardpaths.html 

I can not reproduce the observation on recent development branches: $HOME/Fritzing is not used, as we use QStandardPaths::DocumentsLocation , which usually results in $HOME/Documents/Fritzing .

KjellMorgenstern, 2019-11-24:
I think https://github.com/haikuports/haikuports/pull/1806 is about assets , not user data. This is discussed in #3454

KjellMorgenstern, 2019-11-24:
Won't fix

closed 2019-11-24 by KjellMorgenstern