fritzing/fritzing-app #3352: Disable partschecker and update functionality at compile time

srakitnican, 2017-09-21:
I can understand updating application or parts can be useful, but when packaging for a distro, it is unwanted since application does not have write permissions to its directories and it is against policy of package managers.

Second issue, libgit2 should not be required in this case since it connects to the Internet. Just including `git2.h` tries to do that at compile time.

Sorry for not sending the patches, I've looked at the code and unfortunately I don't know how to do it.

KjellMorgenstern, 2019-07-18:
I think what you are looking for is in tools/linux_release_script/release.sh

KjellMorgenstern, 2019-12-11:
@srakitnican Can you clarify about libgit2 ? How would including git2.h connect to the internet at compile time?

If you speak about runtime notifications: Fritzing will show an notice about a new version once, and then stay quiet about it if the user closes the notification.

As this there was no reaction after "bumping" the issue five months ago, I'll close it for now. Please let me know if it is still relevant.

closed 2019-12-11 by KjellMorgenstern