fritzing/fritzing-app #3353: Ui size windows 10

25pinjo15, 2017-09-21:
The ui is too big on high dpi display. 
Have a 10.2 inch 1920x1200 pixel and the windows scali g set to 150 ( perfect fit for everything) but fritzing is too big. Very not usable. Tryed the disable high dpi thing in the compatibility tab under propriety but nothing work

25pinjo15, 2017-09-21:
[image]

smarbos, 2018-04-26:
Im having the same issue on Linux Mint Sarah (18)
fritzing-0.9.3b.linux.AMD64

justusenumbers, 2018-11-26:
Had the same issue as previously described (Using laptop with Solus Budgie 3.9999). The window size went back to normal once i unplugged the external monitor and restarted Fritzing. It retains proper size even after plugging in the external monitor again. Hope this helps :)

KjellMorgenstern, 2019-04-29:
There is an issue with detecting the right display size. I could reproduce the problem when using Solus, and switching from an 4k to an FullHD screen. Probably the my screen configuration was set to the wrong dpi in x windows, but other applications don't suffer from this. Disabling retina support in the code solved the issue, but will cause problems on retina displays.

KjellMorgenstern, 2021-07-13:
No reports on this for 0.9.4 and 0.9.6 anymore, closing. Please open a new issue with exact information about Operating System, Screen setup and Fritzing version if you encounter this.

closed 2021-07-13 by KjellMorgenstern