fritzing/fritzing-app #3358: workspace is too large for screen and won't allow adjustment

stmaus, 2017-10-16:
The workspace will not scale to fit the screen resolution. Running Fritzing 0.9.3 on Ubuntu 16.04LTS.

Intrinsically-Sublime, 2017-10-17:
There is a good chance it is your scaling in Ubuntu. I run Ubuntu Gnome and by default the scaling is set to 1.2. In Ubuntu Gnome it is in the tweak tool under fonts (It scales more than the fonts). If that does not work I would suggest asking on the support forum http://forum.fritzing.org/

grince, 2017-12-13:
My scaling is at 1.0 and it is bad (see issue 3371 with a screenshot)

Intrinsically-Sublime, 2017-12-13:
Have a read through https://github.com/fritzing/fritzing-app/issues/3221
You can try starting Fritzing with QT scaling turned off.
>QT_AUTO_SCREEN_SCALE_FACTOR=0 ./Fritzing

failiz, 2021-11-08:
Duplicate of #3221