fritzing/fritzing-app #3361: Adding config options to remove extra launchers

albfan, 2017-10-29:
Using 0.9.3

After run fritzing and don't find core parts look into fritzing executables

```
$ which Fritzing
/usr/bin/Fritzing
$ which fritzing
/usr/bin/fritzing
```
Looking inside Fritzing

```
$ cat $(which Fritzing)
#!/bin/sh

cd /usr/share/fritzing/parts
exec /usr/bin/fritzing
```

Is it possible to add a config path in preferences? so there's only a run script, and you can configure where to look fork parts

Don't know exactly how config files are stored on fritzing but a simple ini file would work

```
$ cat ./config/fritzing
[parts]
path = /usr/share/fritzing/parts
```

Let me know if you find this interesting, to provide a PR (as I said maybe the store engine is different, please add some hints) and if it can be expanded to more desirable config options

KjellMorgenstern, 2019-11-13:
On linux systems, Fritzing uses $HOME/.config/Fritzing/Fritzing.conf to store settings.
It would be interesting to add a option to the CLI to set a custom parts folder, for example to allow global with a writeable, shared parts folder.

I don't think it make sense to remove the launcher scripts, if that is what you meant?

albfan, 2019-11-14:
`$HOME/.config/Fritzing/Fritzing.conf` looks nice.

The subject for issue doesn't explain what I want, but only one of the needs, let me rephrase it

albfan, 2019-11-14:
I see https://fritzing.org/learning/tutorials/creating-custom-parts/

but I'm not sure where those new parts ends. 

the whole idea is to load parts base on `Fritzing.conf` and avoid to `cd` to locate them.

Maybe that could allow to run Fritzing with a specific set of parts easily.

KjellMorgenstern, 2019-11-15:
The new part is stored in  "My Parts". 
In the Inspector, you can click the file names, and the folder should open, no manual directory search required.

[image]

The parts will be created in $HOME/Documents/Fritzing/parts

albfan, 2019-11-15:
Nice then. The script cd step shouldn't be needed but is not a problematic thing at all.

Thanks for taking time to explain this, really appreciated.

closed 2019-11-15 by albfan

KjellMorgenstern, 2019-11-15:
Taking a note to highlight the clickable items in inspector on mouseover events.