fritzing/fritzing-app #3364: Fritzing v.0.9.3 crash when saving project file. Results into project lost (0KB file size).

tobiashochguertel, 2017-11-11:
Operation System: Mac OS X High Sierra

Some times the Save process is successfully and sometimes not. When it's not successfully Fritzing crash's and I lost my fritzing project... It happens common so it's annoying.

Is there an debug or development build which I should try - which produces better debug informations for you?

```
Process:               Fritzing [10642]
Path:                  /Applications/Fritzing.app/Contents/MacOS/Fritzing
Identifier:            org.fritzing.Fritzing
Version:               0.9.3 (bda4e)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Fritzing [10642]
User ID:               501

Date/Time:             2017-11-11 22:06:02.978 +0100
OS Version:            Mac OS X 10.13.1 (17B48)
Report Version:        12
Bridge OS Version:     3.0 (14Y618b)
Anonymous UUID:        1AF7100E-3F3C-0E65-09D5-D2B34EED3029

Sleep/Wake UUID:       B5D6846E-5213-4C4F-80D4-2412C6D74805

Time Awake Since Boot: 86000 seconds
Time Since Wake:       63000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x00007ffee7dc5f98
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0x7ffee7dc5f98:
    MALLOC_TINY            00007f95ea000000-00007f95ea200000 [ 2048K] rw-/rwx SM=PRV  
--> STACK GUARD            00007ffee45c6000-00007ffee7dc6000 [ 56.0M] ---/rwx SM=NUL  stack guard for thread 0
    Stack                  00007ffee7dc6000-00007ffee85c6000 [ 8192K] rw-/rwx SM=PRV  thread 0

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.apple.AppKit              	0x00007fff2c6135d3 _screenForRect.llvm.FE30BCA4 + 56
1   com.apple.AppKit              	0x00007fff2bda9ddc -[NSWindow _setFrameCommon:display:stashSize:] + 1362
2   com.apple.AppKit              	0x00007fff2bda987d -[NSWindow _setFrame:display:allowImplicitAnimation:stashSize:] + 222
3   com.apple.AppKit              	0x00007fff2bda9798 -[NSWindow setFrame:display:] + 67
4   com.apple.AppKit              	0x00007fff2bea80e0 -[NSWindow setFrame:display:animate:] + 147
5   libqcocoa.dylib               	0x000000010b2d44ca 0x10b2c4000 + 66762
6   libqcocoa.dylib               	0x000000010b2d4142 0x10b2c4000 + 65858
7   org.qt-project.QtGui          	0x00000001088c741f QWindow::setGeometry(QRect const&) + 1071
8   org.qt-project.QtWidgets      	0x000000010828b476 QWidgetPrivate::setGeometry_sys(int, int, int, int, bool) + 1510
9   org.qt-project.QtWidgets      	0x000000010828ae52 QWidget::move(QPoint const&) + 354
10  org.fritzing.Fritzing         	0x0000000107841acf FileProgressDialog::resizeEvent(QResizeEvent*) + 127
11  org.qt-project.QtWidgets      	0x000000010828e2b8 QWidget::event(QEvent*) + 2168
12  org.qt-project.QtWidgets      	0x000000010824eb76 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 294
13  org.qt-project.QtWidgets      	0x00000001082514c0 QApplication::notify(QObject*, QEvent*) + 5648
14  org.fritzing.Fritzing         	0x00000001076747fd FApplication::notify(QObject*, QEvent*) + 29
15  org.qt-project.QtCore         	0x000000010923e3f4 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 164
16  org.qt-project.QtWidgets      	0x00000001082b04df 0x10823e000 + 468191
17  org.qt-project.QtWidgets      	0x00000001082aea94 0x10823e000 + 461460
18  org.qt-project.QtWidgets      	0x000000010824eb76 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 294
19  org.qt-project.QtWidgets      	0x00000001082514c0 QApplication::notify(QObject*, QEvent*) + 5648
20  org.fritzing.Fritzing         	0x00000001076747fd FApplication::notify(QObject*, QEvent*) + 29
21  org.qt-project.QtCore         	0x000000010923e3f4 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 164
22  org.qt-project.QtGui          	0x00000001088bd313 QGuiApplicationPrivate::processGeometryChangeEvent(QWindowSystemInterfacePrivate::GeometryChangeEvent*) + 339
23  org.qt-project.QtGui          	0x00000001088a092e QWindowSystemInterfacePrivate::handleWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) + 158
24  libqcocoa.dylib               	0x000000010b2db72a 0x10b2c4000 + 96042
[... 1013 lines cut ...]
USB Device: USB 3.0 Bus
USB Device: USB2.0 Hub
USB Device: Hub
USB Device: Hub
USB Device: Samson Meteor Mic
USB Device: Hub
USB Device: Hub
USB Device: USB 2.0 Hub
USB Device: USB 2.0 Hub
USB Device: USB 2.0 Hub
USB Device: USB Receiver
USB Device: USB2.0-Serial
USB Device: USB-C Digital AV Multiport Adapter
USB Device: iBridge
USB Device: USB 2.0 BILLBOARD
USB Device: USB3.0 Hub
USB Device: Hub
USB Device: Hub
USB Device: Hub
USB Device: Elements 25A2
USB Device: Hub
Thunderbolt Bus: MacBook Pro, Apple Inc., 22.3
Thunderbolt Bus: MacBook Pro, Apple Inc., 22.3

```

peleshenko, 2018-01-27:
The same story on Windows 10

84ace, 2018-04-29:
I have the same problem on the latest OSX

dkadish, 2018-11-30:
Same, OSX 10.14.1, Fritzing 0.9.3b

dkadish, 2018-11-30:
Possible duplicate of #1875 
Should be high priority as it results in the loss of all data.

KjellMorgenstern, 2019-07-04:
Debug builds of the most recent version are available now:
https://github.com/fritzing/fritzing-app/releases/tag/CD-268

KjellMorgenstern, 2019-12-11:
Updated debug builds of the most recent version are available:
https://github.com/fritzing/fritzing-app/releases/tag/CD-506

KjellMorgenstern, 2019-12-11:
It would also help to attach a fritzing project ( last saved state before the crash) and describe your actions:
* How long did you work since last time you saved?
* Which view did you work in PCB / Schematic / Breadboard?
* Did you use the autorouter (there was a known bug that could cause any kind of crashes, which is fixed now) ?

KjellMorgenstern, 2021-06-24:
I think the issue has not been reported for 0.9.4 and beyond.

closed 2021-06-27 by KjellMorgenstern