fritzing/fritzing-app #3367: Fritzing doesn't show parts

Abugbilla, 2017-11-17:
Whenever I open fritzing it doesn't give me the parts bin. Instead an error occurs. This is the error I'm talking about
[image]
[image]
[image]
[image]

any attempt to update it does not work either

Intrinsically-Sublime, 2017-11-22:
Are you running the binary download from Fritzing.org? I know when you guild from source or install from a repository this can happen.

sintetico82, 2017-11-22:
I installed it from Ubuntu repository. I will try to download the binary.

Abugbilla, 2017-11-23:
@Intrinsically-Sublime I downloaded it from fritzing.org

Intrinsically-Sublime, 2017-11-23:
If the parts update is interrupted during download it can corrupt the database which requires a complete reinstall. To do a reinstall you must manually delete the hidden Fritzing directory (make a backup first if you have custom parts and sketches already). Then delete the Fritzing program and redownload it. Once you have started Fritzing you may have to wait a long time while it checks for part updates and downloads them. Fritzing will be unresponsive during that time so you just have to be patient and wait until it completes.

Abugbilla, 2017-11-23:
I did that and everything now seems to work fine. Thank you so much for your help. Much appreciated

closed 2017-11-23 by Abugbilla