fritzing/fritzing-app #3368: search

rees52, 2017-11-30:
i am not able to find the search option in new version of fritzing 0.9.3 b

Intrinsically-Sublime, 2017-12-03:
For some reason it sometimes moves to the bottom of the list of parts bins. If you scroll to the bottom of the bins (mine, core, Arduino, Sparkfun,,,,,,,) you should find it at the bottom and then you can drag it back to the top of the list.

gromeo, 2017-12-21:
Hi,

I've also got this issue; I've installed Fritzing 0.9.3b on two computers, one has win 7, the other one has win 10 with latest updates.
On both PC's I've got this issue: the Search bin appears, but with no field in it so I cannot actually search. The debugging info says "loading bin search".
I've noticed that at every startup the program creates a new copy of the "Mine" and "Search" bins and places them at the bottom of the bin's list, so I've ended up with dozens of copies of the two bins. Please note that only the bins in the program are duplicated: no new my_parts.fzb or search.fzb files are created.
It looks like that a part of the problem is that at the startup the program "thinks" that the two bins do not exist and recreates them (not the files, only the icons in the list).

gromeo, 2018-01-06:
I have fixed this issue a few days ago. A new release is needed to get the fix or you can compile the source code.

KjellMorgenstern, 2019-10-12:
I think Giovannis fix is b452175, which is integrated in 0.9.4.

closed 2019-10-12 by KjellMorgenstern