fritzing/fritzing-app #3371: Does not work on Debian 9.2 ( Stretch ) properly

quirxi, 2017-12-04:
Operating system: Linux debian 4.9.0-3-amd64 #1 SMP Debian 4.9.30-2+deb9u5 (2017-09-19) x86_64 GNU/Linux
Debian Version: 9.2 (Stretch)

First I tried to install Fritzing via "apt-get install Fritzing" but the installed program could not install and find the parts for whatever reason. So I decided to download it directly from fritzing.org.
After calling the install_fritzing.sh I wanted to run fritzing but got following error message:
```
# ./Fritzing 
/opt/fritzing/fritzing-0.9.3b.linux.AMD64/lib/Fritzing: error while loading shared libraries: libssl.so.1.0.0: cannot open shared object file: No such file or directory
```
So I installed libssl1.0.0 ( which is the oldest version i could install - I had already installed the newer ones ).
This time Fritzing started but I got following error messages:
```
# ./Fritzing 
Qt: Session management error: Authentication Rejected, reason : None of the authentication protocols specified are supported and host-based authentication failed
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_num_locks
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_set_id_callback
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_set_locking_callback
qt.network.ssl: QSslSocket: cannot resolve ERR_free_strings
qt.network.ssl: QSslSocket: cannot resolve EVP_CIPHER_CTX_cleanup
qt.network.ssl: QSslSocket: cannot resolve EVP_CIPHER_CTX_init
qt.network.ssl: QSslSocket: cannot resolve sk_new_null
qt.network.ssl: QSslSocket: cannot resolve sk_push
qt.network.ssl: QSslSocket: cannot resolve sk_free
qt.network.ssl: QSslSocket: cannot resolve sk_num
qt.network.ssl: QSslSocket: cannot resolve sk_pop_free
qt.network.ssl: QSslSocket: cannot resolve sk_value
qt.network.ssl: QSslSocket: cannot resolve SSL_library_init
qt.network.ssl: QSslSocket: cannot resolve SSL_load_error_strings
qt.network.ssl: QSslSocket: cannot resolve SSL_get_ex_new_index
qt.network.ssl: QSslSocket: cannot resolve SSLv2_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv3_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv23_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv2_server_method
qt.network.ssl: QSslSocket: cannot resolve SSLv3_server_method
qt.network.ssl: QSslSocket: cannot resolve SSLv23_server_method
qt.network.ssl: QSslSocket: cannot resolve X509_STORE_CTX_get_chain
qt.network.ssl: QSslSocket: cannot resolve OPENSSL_add_all_algorithms_noconf
qt.network.ssl: QSslSocket: cannot resolve OPENSSL_add_all_algorithms_conf
qt.network.ssl: QSslSocket: cannot resolve SSLeay
qt.network.ssl: QSslSocket: cannot resolve SSLeay_version
qt.network.ssl: QSslSocket: cannot call unresolved function CRYPTO_num_locks
qt.network.ssl: QSslSocket: cannot call unresolved function CRYPTO_set_id_callback
qt.network.ssl: QSslSocket: cannot call unresolved function CRYPTO_set_locking_callback
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function sk_num
libpng warning: iCCP: known incorrect sRGB profile

(Fritzing:2472): GConf-WARNING **: Client failed to connect to the D-BUS daemon:
Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

and closing the app gives me following error messages:

qt.network.ssl: QSslSocket: cannot call unresolved function CRYPTO_num_locks
qt.network.ssl: QSslSocket: cannot call unresolved function CRYPTO_set_id_callback
qt.network.ssl: QSslSocket: cannot call unresolved function CRYPTO_set_locking_callback
qt.network.ssl: QSslSocket: cannot call unresolved function ERR_free_strings
```
When I run the programm as root *and* normal user Fritzing starts in a huge window that I only can resize horizontally but not vertically thus making the programm practically unusable: 
[image]

When I start Fritzing as normal user I get several error messages that it is not able to find certain parts and one is not able to find them in the application afterwards:
[image]
[image]
[image]

Sharan123, 2017-12-06:
Same here on Kali (Debian based system) the problem goes away when u reinstall fritzing, fritzing-data and fritzing-parts and remove the /usr/share/fritzing folder. But then u cannot update the parts via help->update parts since u get an error that u need to give permissions to /usr/share/fritzing/parts folder.

When i do i get all the errors back that u posted.

Intrinsically-Sublime, 2017-12-07:
Have either of you tried following the instructions on the Fritzing download page? http://fritzing.org/download/ It never says to run install_fritzing.sh script. Just download the binary, unzip it and run it directly. No install needed and the install may be what is breaking things. Also try doing it from your user directory (home directory) not in /opt or /usr etc.

quirxi, 2017-12-07:
Yes - I have tried it any way I could think of before I reported the error here.
In fact the install_fritzing.sh script only installs the fritzing mime type and icons but does not mess with the fritzing installation in any other way. The directory from where I start the program should not be a problem as long as the permissions are set properly. 
What is a bit strange for me is that the developers have linked the libraries dynamically to the Fritzing executable but then provided the libqt5*.so libraries with the download package. I wonder why they have not linked it statically then ? They also seemed to have forgotten to provide the package libssl.so.1.0.0 (and maybe the libcrypto too?) or they just assume that its installed on any linux distribution anyway. But libssl.so.1.0.0 is a bit outdated now - so it cannot be assumed ...
I would either link it fully statically - so that there were no dependencies left or make a proper installation package ( like a .deb package) that resolves all dependencies as libqt5 and libssl. 
That are just my "2 cents" and I am not a total expert in this field..  :P

$ file Fritzing 
Fritzing: ELF 64-bit LSB executable, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.24, BuildID[sha1]=29c0a4472b77c8387ae10596579ce3710302e3a4, not stripped

# ls -a /usr/lib/x86_64-linux-gnu/libssl*
/usr/lib/x86_64-linux-gnu/libssl.so.1.0.0  
/usr/lib/x86_64-linux-gnu/libssl.so.1.0.2  
/usr/lib/x86_64-linux-gnu/libssl.so.1.1
/usr/lib/x86_64-linux-gnu/libssl3.so  

#ldd Fritzing  
	linux-vdso.so.1 (0x00007ffebf2a2000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fe56c5d8000)
	libgit2.so.23 => /opt/fritzing/fritzing-0.9.3b.linux.AMD64/lib/./libgit2.so.23 (0x00007fe56c2b3000)
	libQt5Svg.so.5 => /opt/fritzing/fritzing-0.9.3b.linux.AMD64/lib/./libQt5Svg.so.5 (0x00007fe56c05f000)
	libQt5PrintSupport.so.5 => /opt/fritzing/fritzing-0.9.3b.linux.AMD64/lib/./libQt5PrintSupport.so.5 (0x00007fe56bdf2000)
	libQt5Widgets.so.5 => /opt/fritzing/fritzing-0.9.3b.linux.AMD64/lib/./libQt5Widgets.so.5 (0x00007fe56b580000)
	libQt5Gui.so.5 => /opt/fritzing/fritzing-0.9.3b.linux.AMD64/lib/./libQt5Gui.so.5 (0x00007fe56ad89000)
	libQt5Xml.so.5 => /opt/fritzing/fritzing-0.9.3b.linux.AMD64/lib/./libQt5Xml.so.5 (0x00007fe56ab4e000)
	libQt5Network.so.5 => /opt/fritzing/fritzing-0.9.3b.linux.AMD64/lib/./libQt5Network.so.5 (0x00007fe56a7ef000)
	libQt5Sql.so.5 => /opt/fritzing/fritzing-0.9.3b.linux.AMD64/lib/./libQt5Sql.so.5 (0x00007fe56a5ab000)
	libQt5SerialPort.so.5 => /opt/fritzing/fritzing-0.9.3b.linux.AMD64/lib/./libQt5SerialPort.so.5 (0x00007fe56a392000)
	libQt5Core.so.5 => /opt/fritzing/fritzing-0.9.3b.linux.AMD64/lib/./libQt5Core.so.5 (0x00007fe569c80000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fe5698fe000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fe5695fa000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fe5693e3000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fe569044000)
	libssl.so.1.0.0 => /usr/lib/x86_64-linux-gnu/libssl.so.1.0.0 (0x00007fe568de3000)
	libcrypto.so.1.0.0 => /usr/lib/x86_64-linux-gnu/libcrypto.so.1.0.0 (0x00007fe5689e7000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fe5687df000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fe5685c2000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/libGL.so.1 (0x00007fe568350000)
	libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007fe5680fd000)
	libgthread-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgthread-2.0.so.0 (0x00007fe567efb000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007fe567be7000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007fe5679d5000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007fe567695000)
	libicui18n.so.56 => /opt/fritzing/fritzing-0.9.3b.linux.AMD64/lib/./libicui18n.so.56 (0x00007fe5671fb000)
	libicuuc.so.56 => /opt/fritzing/fritzing-0.9.3b.linux.AMD64/lib/./libicuuc.so.56 (0x00007fe566e43000)
	libicudata.so.56 => /opt/fritzing/fritzing-0.9.3b.linux.AMD64/lib/./libicudata.so.56 (0x00007fe565460000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fe56525c000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fe56c7f2000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007fe565032000)
	libxcb-dri3.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-dri3.so.0 (0x00007fe564e2f000)
	libxcb-present.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-present.so.0 (0x00007fe564c2c000)
	libxcb-sync.so.1 => /usr/lib/x86_64-linux-gnu/libxcb-sync.so.1 (0x00007fe564a25000)
	libxshmfence.so.1 => /usr/lib/x86_64-linux-gnu/libxshmfence.so.1 (0x00007fe564823000)
	libglapi.so.0 => /usr/lib/x86_64-linux-gnu/libglapi.so.0 (0x00007fe5645f4000)
	libXdamage.so.1 => /usr/lib/x86_64-linux-gnu/libXdamage.so.1 (0x00007fe5643f1000)
	libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007fe5641eb000)
	libX11-xcb.so.1 => /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007fe563fe9000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007fe563dc1000)
	libxcb-glx.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-glx.so.0 (0x00007fe563ba6000)
	libxcb-dri2.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-dri2.so.0 (0x00007fe5639a1000)
	libXxf86vm.so.1 => /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1 (0x00007fe56379b000)
	libdrm.so.2 => /usr/lib/x86_64-linux-gnu/libdrm.so.2 (0x00007fe56358b000)
	libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007fe563382000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007fe56310f000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007fe562f0b000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007fe562d05000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007fe562aef000)

grince, 2017-12-13:
I have the same problem - when I start Fritzing (Debian-Package), it creates a huge GUI but it is not useable, see screenshot.
[image]

plazmax, 2018-05-18:
same as ubuntu 18.04, install from ubuntu repository
[image]

mvdiogo, 2018-06-26:
for error ./Fritzing: error while loading shared libraries: libssl.so.1.0.0: cannot open shared object file: No such file or directory on Debian 9 i just
sudo cp /usr/local/lib/warsaw/libcrypto.so.1.0.0 /usr/lib
and
sudo cp /usr/local/lib/warsaw/libssl.so.1.0.0 /usr/lib
and on the lib folder of fritzing
./Fritzing

openelearning, 2018-08-18:
Hello,
On debian 9.5 when I install firtzing with sudo apt-get install fritzing (I have already installed with the tgz files before, but after migrating OS this didn't not work) I had after message error that fritzing could not find some files : 
$ fritzing
QIODevice::read (QFile, "[...]//fritzing-0.9.3b.linux.AMD64/bins/core.fzb"): device not open
QIODevice::read (QFile, "[...]/fritzing-0.9.3b.linux.AMD64/bins/contribParts.fzb"): device not open
QIODevice::read (QFile, "[...]//fritzing-0.9.3b.linux.AMD64/bins/core.fzb"): device not open
QIODevice::read (QFile, "[...]//fritzing-0.9.3b.linux.AMD64/bins/contribParts.fzb"): device not open
 and I have a message error as show in some screenshots of this thread (.../core.fbz)
Installing fritzing-parts with synaptic solved the problem for me.

Hope that helps

plazmax, 2018-10-26:
> Hello,
> On debian 9.5 when I install firtzing with sudo apt-get install fritzing (I have already installed with the tgz files before, but after migrating OS this didn't not work) I had after message [...]

thank you ubuntu 18.10 its solve problem

h3ct0rjs, 2018-12-04:
Hi, 
Whats the proper method to solve this issue on Debian  Gnu/linux system

sjfloat, 2019-02-18:
I'd also be interested in knowing.

HaraldRau, 2019-02-18:
I had the same problem. Now I use a portable installation my folder /home/user/portable/fritzing/!
It works now!
For download the zip-file:
[http://fritzing.org/download/](http://fritzing.org/download/)

sjfloat, 2019-02-18:
Yes. I can think of a workaround or two, a symlink probably being the
simplest. I just thought they may want to fix the bug.

croutor, 2019-06-05:
I was also having this issue:
`./Fritzing                                 
/home/croutor/stow/fritzing-0.9.3b.linux.AMD64/lib/Fritzing: error while loading shared libraries: libcrypto.so.1.0.0: cannot open shared object file: No such file or directory
`
seems that the library pulling libssl1.0.0 is lib/libgit2.so.23 from the Fritzing installation. libgit2.so.24 is installed on my system in /usr/lib/x86_64-linux-gnu
So I've just made the following workaround:
Create a symbolic link:
`
cd /usr/lib/x86_64-linux-gnu
sudo ln -s libgit2.so.24 libgit2.so.23
`
Deleting the libgit2.so.23 that came with the Fritzing package:
`
rm /home/croutor/stow/fritzing-0.9.3b.linux.AMD64/lib/libgit2.so.23
`
And now Fritzing starts properly. Hope this helps.

Symbian9, 2019-09-17:
Just tested `0.9.4` pre-release build on Debian 9.x and it failed due to missing `libssl.so.1.0.0` and `libcrypto.so.1.0.0`.

- https://github.com/fritzing/fritzing-app/releases/download/CD-346/fritzing-32ab1e863c633bbd6bd926dfcc51f14b9b9d50e6-debug-continuous-346.bionic.linux.AMD64.tar.bz2

Also, @probonopd's *AppImage* build of `0.9.3b` has same issue:

- https://bintray.com/probono/AppImages/Fritzing

Found that there is related thread on *servo* tracker and seems like its devs solved such issue in some way:

- https://github.com/servo/servo/issues/21778

probonopd, 2019-09-21:
Another instance of OpenSSL breakage due to not following semantic versioning, see https://github.com/openssl/openssl/issues/5042.

KjellMorgenstern, 2019-09-22:
The openssl versions of those various operating systems (linux versions) simply are incompatible. We do not have the resources to build a binary for each and every operating system. This is quite normal, few Open Source projects are able or willing to do this. 

However, we have binaries compiled with different versions of libssl.
The heroku binary might fit better for older libssl versions (e.g. 1.0.0), and the bionic and fedora 30 binary fit better for newer versions.

In any case, if you want to use Fritzing on a OS different from the ones listed in our official releases, you have to compile yourself, or use a 3rd party binary.

If you want to provide support for any operating system, or formats like AppImage, Snap, or other,  your are welcome. Please get in touch with Fritzing (open a ticket), and we can see what we can do.

closed 2019-09-22 by KjellMorgenstern

Symbian9, 2019-09-22:
> We do not have the resources to build a binary for each and every operating system.

Use Travis CI.

KjellMorgenstern, 2019-09-22:
We already do. And docker, of course.

Symbian9, 2019-09-22:
> for each and every operating system

To be honest, there only 4 packages really needed:

- for Linux - *AppImage* (x86_64);
- for Mac - *DMG*;
- for Windows - EXE and ZIP (portable EXE, optional).