fritzing/fritzing-app #3372: Clean up the open issues.

Intrinsically-Sublime, 2017-12-05:
I have been going through the open issues to see if I can help with anything and I am seeing many old and possibly fixed issues.

Starting with the critical issues I find the following that should possibly be closed now.
https://github.com/fritzing/fritzing-app/issues/3308 and https://github.com/fritzing/fritzing-app/issues/3293 Look to be upstream or in the way Fritzing is packaged for Mac OS. Starting Fritzing from the launcher causes the parts update to fail but it works if Fritzing is started from the command line.

https://github.com/fritzing/fritzing-app/issues/3304 Looks to be a windows 10 pop-up dialog issue where you can not see the dialog behind the Fritzing window. Sounds like it happens for many programs and not just Fritzing.

https://github.com/fritzing/fritzing-app/issues/2939 and https://github.com/fritzing/fritzing-app/issues/2961 and https://github.com/fritzing/fritzing-app/issues/2871 and https://github.com/fritzing/fritzing-app/issues/2919 Are from 2014 and looks to have been fixed.

High priority issues that should possibly be closed.
https://github.com/fritzing/fritzing-app/issues/3223 I believe this is not a bug but intentional as per my comment in the issue. You can not use double row headers in a Breadboard without causing shorts.

https://github.com/fritzing/fritzing-app/issues/2978 and https://github.com/fritzing/fritzing-app/issues/2623 and https://github.com/fritzing/fritzing-app/issues/2618 and most any of the issues marked as imported are from 2014 and looks to have been fixed. Also a lot of imported ones are only links to the old repo which no longer exists and ends up re-directing to the Fritzing home page.

https://github.com/fritzing/fritzing-app/issues/2666 No longer applies because the entire Gerber export has been re-written.

https://github.com/fritzing/fritzing-app/issues/2624 Is from 2014 and is no longer relevant. 

There seem to be too many no longer valid issues to list here and they just need to be cleaned up and any that are still valid that get closed can be reopened when needed. Once the issues have been cleaned up it will be much easier to attack what is left.

nopdotcom, 2017-12-05:
#3339 might help with OS X command line vs Finder launches.

KjellMorgenstern, 2019-05-03:
All issues / PRs mentioned here have been closed or traiged for 0.9.4 / 0.9.5, see also https://github.com/fritzing/fritzing-app/projects . Thanks a lot for the summary.

closed 2019-05-07 by KjellMorgenstern