fritzing/fritzing-app #3373: Download from github (linux)

LyzardKing, 2017-12-05:
Hi!
I'd like to add fritzing to ubuntu-make (https://github.com/ubuntu/ubuntu-make), a tool that helps with downloading and installing the necessary links and files (the menu launcher).
It was discussed a while back in the forum, but I'd need to download from https or to have a checksum (from https).
Would it be possible to put the downloadable tarfile on the github releases page?

Intrinsically-Sublime, 2017-12-05:
Maybe I am not understanding correctly but there is a zip and a tar.gz on the release page for each release as well as the link to the Fritzing download page.

LyzardKing, 2017-12-05:
The zip and tar on github are source code.
What I'd need is the compiled version (that is available on the download page) via a secure connection (so https).
If that's not possible, at least having a checksum distributed via https would be good.

LyzardKing, 2017-12-05:
Also, ubuntu-make would download the latest version, add it to the PATH and add a menu launcher.
There things would have to be done manually with the download from the website

Intrinsically-Sublime, 2017-12-05:
I am only a user trying to help so I can not make the changes but hopefully one of the maintainers can help out.

LyzardKing, 2017-12-05:
@Intrinsically-Sublime It's better to be clear in a github issue, now it's better. Thanks! :)

LyzardKing, 2018-01-05:
@aknoerig If you remember there was a brief discussion in the forum a while back regarding this, but that went nowhere.
Would it be possible now to get either a checksum or https for the download?
You could load the tarfile in the github releases page..

KjellMorgenstern, 2019-07-04:
https download of linux builds from github releases is available, and should be running on ubuntu 16.04 - 19.10.

LyzardKing, 2019-07-04:
Great news @KjellMorgenstern !
It's good to see the activity on fritzing ramping up again.
Will there be both stable "latest" releases and prereleases  on github?
When do you think the next release will be available?

KjellMorgenstern, 2019-07-04:
@LyzardKing We have a continuous delivery, currently it is set to create draft releases. From those I publish pre-releases. If feedback for the current [pre-release](https://github.com/fritzing/fritzing-app/releases/tag/CD-268) is good, a release will follow soon, so it is a matter of weeks. So yes, both will be on github.

KjellMorgenstern, 2019-07-08:
Closing this for now, let me know if you need anything more for the umake distribution.

KjellMorgenstern, 2019-07-30:
Ups... the current build depends on libssl.so.1.0.0.
I think that would be when releasing via ubuntu-make?

LyzardKing, 2019-07-30:
Yes, that would be an issue on newer versions of ubuntu...
Can the libssl dependency be "easily" updated?

KjellMorgenstern, 2019-07-30:
Updating would be a change in the build environment, no code change (e.g. I work on 19.04 without changes to the repo) . If ubuntu-make does the build anyway, there shouldn't be a problem. If we should provide a binary, then I'd have to setup a new target that builds on a more recent ubuntu.

LyzardKing, 2019-07-30:
Ubuntu-make Is binary only
But 19.04 should be good, it doesn't have the old version of libssl...I think

LyzardKing, 2019-07-30:
19.04 has libssl1.1
That should be fine

LyzardKing, 2019-08-06:
on the previous versions (16.04, 18.04 and 18.10) the libssl version is 1.0
So with the previous releases it won't work..

LyzardKing, 2019-08-06:
@KjellMorgenstern 
In the travis file It seems that it's built on a xenial docker image..
It might be good to add an 18.04 version, so it's  possible to have a "xenial" release with libssl1.0
and a "bionic" release with libssl1.1

KjellMorgenstern, 2019-08-06:
I added a second target (see the develop branch). 
So there are nowls -l two builds, using docker. One uses an image based on 
conan/gcc5, which uses ubuntu:xenial
and the other
conan/gcc8, which uses ubuntu:bionic

Both install  libssl via
 apt-get -qq install -y --no-install-recommends    libssl-dev=1.* 

The build with image conan/gcc8 (bionic) also works on disco.

LyzardKing, 2019-08-06:
I'll add fritzing as soon as the first release with this new target is out!
Thanks!

LyzardKing, 2019-08-31:
Hi @KjellMorgenstern !
I'm adding the final touches now, the download works perfectly..
I added as dependencies only libssl, libqt5serialport5,libqt5svg5...if there are other ones that might not be installed in a version of ubuntu let me know (At the moment I only tested on kde neon...)
Just one question, when I launch I get an error regarding an error with the parts...am I correct in thinking that the github release of fritzing does not include them?

LyzardKing, 2019-08-31:
[image]

LyzardKing, 2019-10-05:
Fritzing depends on libgit2.so.26, while eoan has verison 27.
I know eoan (ubuntu 19.10) is not released yet, but is it not possible to depend on the .so file and not a specific version of  the library?

KjellMorgenstern, 2019-10-05:
What is the output if you run `ldd Fritzing` (with the binary, not the script) ? Are there any libraries missing?

LyzardKing, 2019-10-05:
The only one missing is: `libgit2.so.26 => not found`
But it should be in the tarfile, right?

KjellMorgenstern, 2019-10-05:
Thanks, I will check whats wrong.

LyzardKing, 2019-10-22:
And fritzing is now added to ubuntu-make!
I think we can close this issue now ;)
Thanks @KjellMorgenstern

KjellMorgenstern, 2019-11-09:
Thanks LyzardKing!

closed 2019-11-09 by KjellMorgenstern