fritzing/fritzing-app #3375: New Gerber Export has multiple bugs.

Intrinsically-Sublime, 2017-12-13:
The new gerber export committed march 31, 2017 https://github.com/fritzing/fritzing-app/commit/48eabe3af3baa7da2ced8383417c792828e29966 and patched December 4, 2017 https://github.com/fritzing/fritzing-app/commit/4a7ed22ed96e64cba969cf1ead3643afa1c0dcc9 to fix the most obvious bug. Has multiple bugs including not being RS274-D compliant.

Issue discussed http://forum.fritzing.org/t/new-gerber-export/5013/

1) It polygonizes everything thus stripping all extended data. Gerbv complains about copper and mask layers not having any apertures defined. This is because it no longer defines pads and flashes.
>Most likely found a RS-274D file...trying to open anywaysMissing apertures/drill sizes...trying to load anyways

2) It produces mirror layers in the gerber output which you do not need (I think you only want these for SVG and PDF output).

3) It is producing paste masks for through hole parts which you do not want.

4) The drill sizes are wrong. (In Fritzing the via has a 0.015748" hole and the gerber drill file says T101C0.026559)

5) The paste mask is larger than the solder mask which is opposite of what you want.

See Fritzing forum for examples and pictures http://forum.fritzing.org/t/new-gerber-export/5013

Until there are people actively working on Fritzing I feel it would be best to move the current head to a New-Gerber branch and revert the changes so we can focus on getting another working release out with all the bug fixes from the last year of commits.

aknoerig, 2017-12-18:
Thanks, I think this is the right way to go until the new style export can be worked on.

failiz, 2023-02-22:
The "new gerber export" has been reverted a long time ago, close?

KjellMorgenstern, 2023-02-22:
Closing in favor of #1457 #1568 #3978 

Fixing those issues might involve repairing or porting some of the patches of the 2016/2017 "new gerber export"

closed 2023-02-22 by KjellMorgenstern