fritzing/fritzing-app #3386: 0.9.3 not finding parts under Ubuntu 17.10

hadmut, 2018-01-02:
Hi, 
I had been using fritzing 0.9.2b under Ubuntu 16.04 so far, no problems. 

Under 17.10 the 0.9.3b version (both the packaged version that comes with ubuntu and the downloaded 64bit-binary from fritzing.org) don't find  any parts. When I start a project, it just shows an empty breadboard and the Inspector menu, but there is not parts library, just three empty white squares. 

Furthermore fritzing seems to have problems keeping the cwd and the library path apart. Under 17.10 the ubuntu maintainer has – for whatever reason – renamed the binary and put a shell script with

#!/bin/sh

cd /usr/share/fritzing/parts
exec /usr/bin/fritzing

put in it's place, obviously trying to fix problems and heaps of error messages that occur when starting the binary directly. 

Fritzing 0.9.3 seems to have some issues with path handling. That needs to be fixed. 

regards

birgersp, 2018-02-08:
+1 wanted to use Fritzing today, but couldn't get it to work

kfowlks, 2018-02-25:
Has there been any thoughts on creating an Ubuntu AppImage for fritzing? This should alleviate the issues like the above.
 
[AppImage](https://github.com/AppImage/AppImageKit/wiki/Creating-AppImages)

Sembiance, 2018-04-04:
I'm having the same problem with 0.9.3b. `Fritzing -f /usr/share/fritzing` doesn't help and neither does `Fritzing -f /usr/share/fritzing/parts` It's like Fritzing just ignores the `-f` property entirely in 0.9.3b

Sembiance, 2018-04-04:
Turns out Issue #3286 covers most of these same problems.

I ended up solving it by doing: `ln -s /usr/share/fritzing/parts/ /usr/share/fritzing-parts`

installfails, 2018-06-13:
Affects Debian, Ubuntu, and related distros.  The problem seems to be that Synaptic does not include fritzing-parts library in the download and install process.  Ugly work-around is to manually download fritzing-parts from your favorite repository and re-run the install process.

XtianDR, 2018-09-13:
I fixed this issue when I installed fritzing in my Ubuntu 18.04. Using a terminal:
sudo apt install fritzing
sudo apt install fritzing-parts
And its works like a CHARM!

techedemic, 2018-11-06:
@XtianDR 

> I fixed this issue when I installed fritzing in my Ubuntu 18.04. Using a terminal:
> sudo apt install fritzing
> sudo apt install fritzing-parts
> And its works like a CHARM!

Same here. Think it should be a linked dependency in `apt install`. Mine is working after doing what you suggested, but I get the odd freeze every now and then.

KjellMorgenstern, 2019-06-29:
fritzing-parts is only suggested on
artful, bionic and cosmic

But it is a fixed dependency on
xenial, disco and eoan, so those should be fine.

closed 2019-06-29 by KjellMorgenstern