fritzing/fritzing-app #3399: Cannot Save Sketch in location outsize my documents

rhj4, 2018-03-01:
I keep my Arduino sketches on a separate drive, but when I try to save my Fritzing file to that location, it says I don't have authority to do so. I am the admin and I added ALLOW ALL to all authenticated users, but that doesn't work.

KjellMorgenstern, 2019-07-07:
Can somebody with a native windows installation (no VM) and two harddrives confirm this issue?

dadul96, 2019-11-01:
I can not reproduce this issue on my system:
- Fritzing version 0.9.3b
- native Windows 10 Pro 64-bit
- two hard drives
- ALLOW ALL for all users
- run as Admin

KjellMorgenstern, 2019-11-01:
@dadul96 I just noticed, maybe with "separate drive", a usb drive or stick was meant. Just to be sure, could you 
* add some (dummy) arduino code in the fritzing code editor, right of the PCB tab
* save that Fritzing file to a usb stick on your native Windows system?

dadul96, 2019-11-01:
@KjellMorgenstern This error still does not occur on my system. (Tested with Arduino dummy code and saved on FAT32 and NTFS formatted USB-sticks)

KjellMorgenstern, 2019-11-01:
Closing this as 'can not reproduce'.
@rhj4 If you still can reproduce this issue, please post screenshots , the exact wording of the error message ('You don't have authority' is a rare wording for error messages), and any detail you think might be helpful.

closed 2019-11-01 by KjellMorgenstern