fritzing/fritzing-app #3402: Sign releases, provide hashsums, set up TLS on fritzing.org

0xBADCA7, 2018-03-24:
Right now there's no safe way of downloading a binary release. Please provide:

- Binary release hashsums (sha256 or even sha512)
- Opportunistically, GPG-sign the releases
- Set up a TLS server on fritzing.org b/c downloads are via HTTP and no hashsums or signatures to be found whether here on GH or on the home page.

0xBADCA7, 2018-03-25:
Partially includes #3393

advayDev1, 2018-07-08:
A few more security requests:
* For the macOS downloads, please use App Signing.  (Also let's the user confidently enable firewall permissions for this app to download parts) 
* Provide the download releases directly via the GitHub release mechanism (at least then the user knows the download is coming securely from GH)

KjellMorgenstern, 2019-05-02:
Problem with this ticket is that it contains multiple feature requests at once. Setting the focus to hashsums.

closed 2019-07-08 by KjellMorgenstern