fritzing/fritzing-app #3404: Duplicate "My Parts" and "Search" bins on Windows

wkorn, 2018-03-30:
I recently noticed that my Fritzing installation creates a new pair of "Search" and "My Parts" bins upon each start so that the toolbar holding the bins gets longer and longer each start. The duplicated bins are appended to the end of the toolbar. Also the Search bin is not usable since no textbox opens.

After investigating this issue I figured out the reason for this behaviour. My Windows installation uses to HDDs and the "Documents" folder was moved to the second disk. Moving it back to C:\User\xyz\Documents fixed the issue. Moving it back to the second disk made the problem reappear. I finally solved it on my system by creating a junction from C-Documents to my second disk's Documents folder (mklink /j). However, Fritzing should cope with Documents folders not residing on the system drive.

I'm using Fritzing 0.9.3b on Windows 10.

KjellMorgenstern, 2019-05-10:
It would be great if someone could take the time and reproduce the issue. 
This might also be a hint to "my parts" and "search" bins disappearing.

KjellMorgenstern, 2019-05-10:
About the requested feature: Fritzing already uses `QStandardPaths::writableLocation(QStandardPaths::DocumentsLocation)` to determine the documents folder. On Windows, this resolves to FOLDERID_Documents which is `%USERPROFILE%\Documents`
See https://docs.microsoft.com/en-us/windows/desktop/shell/knownfolderid .

This basically means that the feature requested (support drive D:) is already supported, as long as the USERPROFILE variable is correctly set. Windows itself does in general not support moving only the documents folder alone.

closed 2019-05-10 by KjellMorgenstern