fritzing/fritzing-app #3408: incorrect auto route 7 segment display

Wolfgang1126, 2018-05-08:
i just made a simple circuit which includes 5 74HC595 shift register and 5 pcs 7 segment displays. Since i sit in Austria i need to buy the 7 segment displays on the local market. The  order of the connections is different to the ones from the fritzing library. Here my problem:
Even though i connected the 7 segment displays correctly on the bread board page the autorouting is ignoring it and creates  wrong connections on the cicuit. E.A. my 7 segment display have the common cathode on 3 and 8. But the circuit autoroutes them to 9. The segments as well are wrong routed.

Any idea what i am doing wrong?

[Mini Scoreboard V2018.zip](https://github.com/fritzing/fritzing-app/files/1984523/Mini.Scoreboard.V2018.zip)

vanepp, 2019-05-04:
Assuming the original poster is still interested in a fix (this issue is a year old now), I can probably provide one. I had a look at the sketch and I think it is likely that there is database corruption as the schematic doesn't appear to match breadboard when partially routed. That will likely break auto routing. The correct answer would be to create a part that matches the seven segment display in use rather than try and modify an existing one. In addition the part in core parts is incorrect (it is missing terminalIds in schematic, although that sholdn't be what is causing the problems in the sketch.) The place to start would be the pinout of the LED displays that the OP wants to use to locate or make a correct part. 

Peter

KjellMorgenstern, 2019-05-04:
@vanepp Is this (missing terminalIds) already fixed in nightlyParts?

vanepp, 2019-05-04:
Not as far as I know. I just noticed it while looking at this sketch and made note it needs to be fixed. Unfortunately its not uncommon there are a fair number of similar issues in core parts.

Peter