fritzing/fritzing-app #3411: swapping failure

installfails, 2018-06-10:
Swapping failure notice on opening the application.  Cannot load parts libraries.
THIS IS AN APPLICATION KILLER!

installfails, 2018-06-13:
This appears to be a problem in Debian Linux and related Linux distros if using Synaptic to install fritzing.  Due to dependency errors the fritzing-parts library is not being downloaded and installed.  
A work-around is to manually download fritzing-parts from your favorite repository and re-run the install process.  Not many will know how to do this, but it did work for me.

KjellMorgenstern, 2019-05-11:
PR #3467 might fix this.

KjellMorgenstern, 2019-12-11:
In case you encounter this via synaptic or other installers, please try the official release 0.9.4 available from https:/fritzing.org .

closed 2019-12-11 by KjellMorgenstern