fritzing/fritzing-app #3418: Project giving every appearance of having died

garthk, 2018-07-14:
I'm concerned Fritzing is an ex parrot:

* We can't register (#3417)
* We can't update parts (#3403)
* The site lacks HTTPS, increasingly a sign of disrepair (#3402, #3393)
* The [Twitter account](https://twitter.com/fritzingorg) hasn't seen a post in 1200 days
* There have been no [blog posts](http://blog.fritzing.org) for a year
* GitHub activity has dried up (last commit 6241875 on March 13)
* Issue maintenance dried up earlier: @Intrinsically-Sublime tried to help in December (#3372) but couldn't do anything without admin rights; we're now at >1K open issues

First up, I'd be delighted to be proven wrong and close this issue.

Just in case, though:

Without help from the previous maintainers, a small team with sufficient time and energy could rescue the app somewhat: set up a new “official” fork repo and web site, build servers, download pages, and certificates, and add sharing mechanisms that don't need `fritzing.org` e.g. `gist.github.com`, `imgur.com`, wherever else people want their stuff. 

With help from the previous maintainers, the team could take over the official repos and, perhaps, the site, but it'd need effort and — for the site — funding. Also, if I were @aknoerig et al I'd be pretty reluctant to hand over the keys without proof of effort by some sane people.

Anyone up for at least getting a shared fork, builds, and downloads working? I'll ping some people who'd committed to forks up to March: @Intrinsically-Sublime, @CommanderPho, @lojoyu, @luzpaz, @vanepp, @silegit, @vandumxx. Tag anyone else you want.

Intrinsically-Sublime, 2018-07-14:
I do not have much to add other than to say there is some life in a new fork of Fritzing based on web technologies. I do not know if it is where all the effort is being put and the reason the the main program is looking to be abandoned. You can read a bit about it on the forum http://forum.fritzing.org/t/fritzing2web-looking-for-contributors/5653 .

I myself have no interest in a web based Fritzing and maintain my own repo with changes that I have felt were needed and I was capable of implementing. It does require compiling to use so it is likely just me using it. https://github.com/Intrinsically-Sublime/fritzing-app/tree/Testing

vanepp, 2018-07-14:
To my mind this "a small team with sufficient time and energy " is the issue. I haven't seen so far seen anyone with the time or expertise to do this successfully and I have been encouraging people in the fritzing forum for some time to join in and help. I've got poorly tested (so far) fixes for a couple of bugs which tells me the code base is complex and it isn't clear that fixing a bug doesn't introduce new ones by things I don't know about the interaction of the various pieces. Without the help of someone who is familiar with the code,  I'm not sure that is possible.  There was agreement some time back that a new release was possible if there was properly tested code available. The web based replacement is being done by one of the original developers (who is also the parts maintainer how does update parts when a proper request is made here on github) but I don't know enough to even get very far on helping with that. I have a development environment (which wasn't trivial to set up) running on Linux, but have not been successful in getting one running on Windows, as there is too little documentation on what the original Windows setup looked like (and the original platform is several years out of date by now). I think a good starting place is to get a publicly documented (started in a fritzing forum post) development environment howto so people can try out (by compiling from source) fixes. I'll admit I haven't done anything on that in a while, looking to see if the web based one will do something useful, but I think that is the appropriate place to start before trying to fork a new repro. We would all be better off with a new release on the official Fritzing site, but to do that we need to be able to create a test environment that works for all three platforms. I can do Linux and Windows (if I ever figure out how) but have no Macs. 

Peter

Mordna, 2018-07-15:
Fritzing won't be abandoned - but the client implemented in c++/qt will. They are working on a js/React-version of a server/client. Also there should be a offline version too later - according to the thread mentioned before.
A server/client implementation is actually a good choice since it allows the usage of Fritzing on low-performance devices like tablets, ultrabooks (the bad ones). Also one **could** potentially add a function which allows the edit of one file from multiple person in near real-time (however, this is only a nice-to-have and it was **not** mentioned yet from them). For this, javascript (with React-framework) is a nice choice, since it is easier than c++, support cross-platform and could potentially be used from the web.
However, this is only my interpretation of their writings, it **could be different**.
Also, I am not exactly a fan of having a server implementation written in javascript due to the (is completely depended on their implementation) potential performance loss (f. ex. vs-code's language server (NodeJS) which is still worse than intellj's (java), at least on my machine) - although Fritzing is more aimed at hobbyists, in which case it makes no difference.
But at least Fritzing is not dead yet. 
I would not recommend forking the repo and adding stuff, but starting again from zero (with the knowledge of this and fritzing-js) - this would require lots of time and a few people. So I don't think that a new and better clone will emerge easily but you are free to prove me wrong.
This is my opinion, so you are free to disagree with me or make your own one.

nopdotcom, 2018-07-15:
I don't see the harm in forking the repo for maintenance while the (independent) rewrite is in progress. I think it would be a good thing.

If you need a Mac user for simple testing, I'm available. I can do builds too, but I'm not a paid Apple developer this year, so I can't sign builds.

StripedMonkey, 2018-07-23:
After trying to install fritzing in order to help a friend out and finding This little gem of unusability:
[image]
I come here and find this, if some of the community is willing to try and maintain a fork and nobody on the official branch is willing to read/do PR work I think its perfectly acceptable to start maintaining a different fork. I'm willing to do some work to help out if @Intrinsically-Sublime and whoever want to work on a fork of fritzing. I too, have little to no use for a mobile version as half the point of *having* this is for situations where I have no internet access (which happens a lot, especially with how my internet is at home.)

vanepp, 2018-07-23:
I don't have a problem with the community doing work, however that said,  I haven't seen any evidence of it happening in a couple of years of trying to spark interest either. It isn't fair to say there has been no work done either. A year or so ago head wouldn't even build anymore, that got fixed and when we discovered that the new gerber code (a huge commit) was in fact broken it got backed off.  Parts updates occur if you submit a correct part via github using the documented procedures (I've done a number of them). It doesn't look like the parts submit forum page is a path any more, but parts updates are possible and encouraged. Your problem above looks like database corruption which usually happens when you interrupt the initial parts update before it completes on first run but is fixable, the fix is available in the fritzing forums.) At the time the gerber code was backed off, the developers said that they were open to  a new release if we came up with a stable tested set of fixes (which is why I don't think a fork is necessary or desirable). That hasn't happened, partly because I have so far been unable to get a build environment working on Windows only on Linux. I have been somewhat lazy in the last 6 months or so but may start beating my head against getting libgit2 running on Windows again. Until we can get people able to get a copy of head (or our fixes if they aren't in head) to test on the most common platform (Windows) I don't think we are going to get far. Also the code base is very complex (perhaps less so if you are already familiar with QT) but with enough work I can usually (but not always) figure it out. So many things are triggered from an action (and thus from a change) it is a challenge to make sure a fix doesn't break something else. Coming at the code cold with no previous developers to provide hints of what may break if you do something progress is going to be slow and hard. That may be what is discouraging people from helping. I'd say at this moment what needs to happen is for me (or someone else!) to figure out and document (I have started this for linux in a forum post in developers) for all three platforms linux (done), Windows (in progress sort of but help welcome) and Mac (shouldn't be hard as it is unix under the covers) but which I don't have. With that we can publish fixes for testing although you have to build from source to do so. I'd welcome more people with time and expertise contributing preferably via the fritzing forum as it has (although that is dying off too) the widest readership and thus the best chance of attracting interested people. 

Peter

StripedMonkey, 2018-07-23:
>I don't have a problem with the community doing work, however that said, I haven't seen any evidence of it happening in a couple of years of trying to spark interest either.

Multiple people within this thread alone have expressed willingness to try and get some stuff tested or worked on projects, especially ones like these, do not get a ton of the attention they deserve. That does not mean however, that progress cannot be made. It simply has to be *able* to happen. In the instance of this project that is not happening, simply because nobody can actually *do* anything. The longer someone has to wait, pester, or email people before shit gets done makes things *not* get done. Nobody has the time to wait months for responses to PRs or bug reports. That doesn't even take development time to do! that just requres someone that can be trusted to *read* Issues.

>Your problem above looks like database corruption which usually happens when you interrupt the initial parts update before it completes on first run but is fixable, the fix is available in the fritzing forums.

Its actually because I installed it from the repository which, for *whatever* reason does not depend on the fritzing-parts package. My comment was more on the insanity of having an error message like that and **hardcoded** paths like `C:/User/jonathan/...` which is completely invalid considering I'm not on Windows nor am I Jonathan. Its a level of laziness and quality in making tools usable for everyone that is frankly disappointing in something that is touted as a modestly professional/hobbyist software by a **ton** of sites.

>I'd welcome more people with time and expertise contributing preferably via the fritzing forum as it has (although that is dying off too) the widest readership and thus the best chance of attracting interested people.

Its *not* a platform for tracking issues that is particularly well suited, especially as the codebase is held on github where there are already tools that are particularly well intergrated with git. I'm probably going to stick with looking primarily at the github simply because, while I'm willing to help, I have other things that I dedicate time to and keeping track of yet another forum is simply impractical for me, especially as classes are starting up soonish.

All I'm saying is that if someone that currently has administrative privileges is willing to dedicate some time to maintaining the *issue* tracker or giving someone who is willing to do so would massively improve the situation, especially for prospective developers who want to help out but see that people who have tried have simply met an empty house.

StripedMonkey, 2018-07-23:
>I don't have a problem with the community doing work, however that said, I haven't seen any evidence of it happening in a couple of years of trying to spark interest either.

Multiple people within this thread alone have expressed willingness to try and get some stuff tested or worked on projects, especially ones like these, do not get a ton of the attention they deserve. That does not mean however, that progress cannot be made. It simply has to be *able* to happen. In the instance of this project that is not happening, simply because nobody can actually *do* anything. The longer someone has to wait, pester, or email people before shit gets done makes things *not* get done. Nobody has the time to wait months for responses to PRs or bug reports. That doesn't even take development time to do! that just requires someone that can be trusted to *read* Issues.

>Your problem above looks like database corruption which usually happens when you interrupt the initial parts update before it completes on first run but is fixable, the fix is available in the fritzing forums.

Its actually because I installed it from the repository which, for *whatever* reason does not depend on the fritzing-parts package. My comment was more on the insanity of having an error message like that and **hardcoded** paths like `C:/User/jonathan/...` which is completely invalid considering I'm not on Windows nor am I Jonathan. Its a level of laziness and quality in making tools usable for everyone that is frankly disappointing in something that is touted as a modestly professional/hobbyist software by a **ton** of sites.

>I'd welcome more people with time and expertise contributing preferably via the fritzing forum as it has (although that is dying off too) the widest readership and thus the best chance of attracting interested people.

Its *not* a platform for tracking issues that is particularly well suited, especially as the codebase is held on github where there are already tools that are particularly well intergrated with git. I'm probably going to stick with looking primarily at the github simply because, while I'm willing to help, I have other things that I dedicate time to and keeping track of yet another forum is simply impractical for me, especially as classes are starting up soonish.

All I'm saying is that if someone that currently has administrative privileges is willing to dedicate some time to maintaining the *issue* tracker or giving someone who is willing to do so would massively improve the situation, especially for prospective developers who want to help out but see that people who have tried have simply met an empty house.

durasj, 2018-09-05:
Regarding the web version, it can be later easily wrapped in the Electron to become an offline desktop application. But creators need to count with that upfront, so they don't rely on servers.

undefinedLogic, 2018-09-12:
Today I was able to download it and use it to make a schematic. Is this a problem that was fixed, or was this never a problem?

StripedMonkey, 2018-09-12:
Not really no, although there are a number of things that are unstable/crash

caseyfw, 2018-10-22:
@undefinedLogic - I installed a couple of days ago on Ubuntu 18.04 from APT and saw issue shown in the "unable to find parts" screenshot above posted by @StripedMonkey. As they mentioned, it's due to the main `fritzing` APT package not depending on the `fritizing-parts` one, which is somewhat of a cockup.

vanepp, 2018-10-22:
Ah, but a package installer issue not a Fritzing issue as installing from the installation tar file on the Fritzing site as recommended includes the parts repo as part of the install. 

Peter

tuxun, 2018-12-08:
appart that it still look like a dead project, on the website and on this github at the first look.

vanepp, 2018-12-08:
At this point I'd have to agree. The forum signup has been dead for 9 months or so (I don't know exactly why, I suspect no one with the time / expertise/access or possibly money to update the captcha code because it has been reported). The Windows distributions will no longer do parts updates due to libgit2 using tls1.1 which github  no longer supports. I tried replacing just the libgit2 dll with a newer one (which works when compiled from source) but it doesn't work and I don't know enough about Windows dlls to know if there is something I can do to fix it. I don't have a debug version of the Fritzing 0.9.3b release code to compare it with. If someone has a working 9.3b / Qt5.6 installation and could try getting a later libgit2 dll to work with the current distribution that would be great. Just replacing the libgit2 dll in the distribution would be a small fix and likely to get done I expect (and if not, then yes the project is dead!) I'm slowly poking at fixing this but in addition the 32 bit open source version of Qt hasn't installed for about the last 6 months either and they appear to have backported a fix to Qt5.6 which breaks Fritzing. That can be fixed by a parts update to remove tspans from parts which is what I'm currently working on to get to the point where a build from the source head works again. At present it builds, but dies after the parts update introduces parts using tspans. The base problem is no one willing to actually do work on the project. I'm carrying on because it does what I want and I can make my own parts and do minor bug fixes even if I can't distribute them for all platforms (I don't have any Macs). Help would be appreciated by all I expect but I'm not all that hopeful. It looks to me that the javascript version has also stalled, there haven't been any updates for some time. While I don't know javascript and thus can't help them I can clean up the parts data base which will help both projects. 

Peter

tuxun, 2018-12-08:
@vanepp and @Intrinsically-Sublime what about 

> I think the reason it is dead is because the creator no longer had time for it and gave up control to a non-profit that ironically wants more money before they will do any more work. The site has a copyright notice of 2010 and the writing implies it hasn’t been updated since then.
http://friends.fritzing.org/ 44
Who own this repo? the website access for put a simple news on it?
Have you seen https://blog.qt.io/blog/2018/11/23/qt-quick-webgl-release-512/ ? :)

Intrinsically-Sublime, 2018-12-08:
@tuxun  As far as I can tell it is all owned by "Friends of Fritzing" which is a German registered non-profit. The repo itself only has 4 people with commit access https://github.com/orgs/fritzing/people
The friends of Fritzing website which seems to be failing now has a list of Menbers if anyone wants to try contacting them about the issues. You can find it on the wayback machine since the site itself is not loading correctly.
https://web.archive.org/web/20171022050520/http://friends.fritzing.org:80/members/

ccoenen, 2018-12-08:
I wonder if this has been taken down on purpose. I've seen the website a few hours ago. And at some point it stopped working. I didn't think much of it at the time, I thought it was my connection.

StripedMonkey, 2018-12-10:
AFAIK the intent was to move towards a browser based app that had at least some cloud component. Personally I disagree with that movement as I really would prefer not to have myself chained to a place with decent internet access. This is especially true as we obviously can't rely on the server-side portion of the application being up forever considering the *current* issues we're having with the project.

If at least a ~~majority~~ good number of us can reach a consensus on how/what (go forward with web-based, qt, etc.)to fork I'd be willing to pitch in to help maintain the project, but I'm certainly not willing to do it on my own. At this point I think that even if we came to the consensus of moving Community development to a fork that an active user will be able to merge PRs into that would be a step in the right direction.

vanepp, 2018-12-10:
@StripedMonkey I am by and large in agreement, if they make it with the javascript version all to the good, but other than parts help I don't know enough to be useful and would rather have a self contained version. However I am willing (if not necessarily that competent) to try and carry on the current project. As I've said before I think we are a long way away from needing a fork (it is however looking like one will be necessary eventually). First we (or I as no one else seems to be doing it) need to get a development environment working and documented so it is easier for others to join in if they choose. The current documentation (such as it is) is out of date and incomplete. I'm somewhat there, after a couple of years of working on it. I have Linux (easy) and Windows (much harder) dev environments running. As noted it hasn't done me much good because Qt has changed and the parts database as it currently stands breaks when the  resulting build does a parts update. Hopefully soon my part checking script will be able to automatically remove the tspans in the parts that have them and fix that by doing updates to the parts data base. Then the build environment (at least for Linux and Windows) can be documented and we will see if there is interest. Hopefully someone on the Mac side can figure out and document a build environment because as noted I don't have a Mac. Given how easy Linux is to get working I don't expect that to be a big issue, Windows is the problem child. Once we have some progress on widely tested (which is going to be a problem I expect) fixes, then we can see if there is interest in making a new release. A year or so ago when the gerber changes were backed out of the development head, there was the suggestion that a new release was possible if there was a stable well tested version available to release but without the gerber code there wasn't much other change, and a chicken/egg problem exists with testing a new version if no one could or would build one and make the result available to others to test. To my eyes nothing has changed in the last year (although I am slowly working on changing that). 

Peter

KjellMorgenstern, 2019-05-02:
This issue was closed because it is outdated.
If you think this issue is still important for a reason, please open.

PS Thanks a lot for the summary, the issues have been triaged.

closed 2019-05-02 by KjellMorgenstern