fritzing/fritzing-app #3419: Problem with swapping mechanism and can't find parts

jwmurphyjr, 2018-08-24:
Upon starting program in Debian 9.5 Stretch with fritzing & fritzing-data 0.9.3b+dfsg-4 & fritzing-parts 0.9.3b-1 installed ,  I get this error

[image]

also, can't read file screw_terminal_2_3.5mm.fzp
[image]

and unable to find following parts
[image]

jwmurphyjr, 2018-08-24:
The program loads after these errors

jwmurphyjr, 2018-08-24:
Just found issue problem, fritzinf-parts folder is now being created when installing through Synaptic Package Manager and Debian repository. Work around is to download program fritzing-0.9.3b.linux.AMD64.tar.bz2 package from http://fritzing.org/download/ uncompressed and copy the fritzing-parts folder to /usr/share folder and the errors go away

ngvjai, 2019-05-04:
@jwmurphyjr  Thank you for your sharing. I solved this problem with your method.

abbyjgeorge, 2019-05-10:
@jwmurphyjr Thanks, that worked!

KjellMorgenstern, 2019-05-11:
PR #3467 might fix this.

akai54, 2019-09-11:
Hello i had the same probleme on kali linux so i just compressed the folder from fritzing website then copied the folder "fritzing-parts" to /usr/share/fritzing it solved the probleme.

closed 2019-12-11 by KjellMorgenstern