fritzing/fritzing-app #3420: Where's my search bar?

dalek-anonymous, 2018-08-26:
0.9.3b lost its search bar.  I reinstalled a few times with restarts and everything I could think of.  I need that feature.

andrebruton, 2018-09-07:
Two weeks and still no search facility. Can't use Fritzing without a search. There needs to be faster fixing or updating the software. I've moved on to DesignSpark...

fesch, 2019-02-03:
My search bar disappeared too ... :-(

juanan0, 2019-04-25:
yo tampoco puedo usar el botón de search.. aparece en gris y no me da la opcion de escribir. en las versiones anteriores si que funciona correctamente

KjellMorgenstern, 2019-05-08:
This seems difficult to reproduce. It looks as if the search bar just vanishes at random. But if anybody has an idea what happened before it vanshes, please post.

KjellMorgenstern, 2019-05-10:
Might be related to issues with the Documents folder (see #3404)

acboother, 2019-07-03:
I’ve had this issue of the search option missing for sometime on a Windows 10 PC. I found that Fritzing files were being stored on my OneDrive now. The search bin search.fzb was there. Noting that the my parts bin was now called my_parts.fzb I renamed search.fbz to my_search.fbz and hey presto a search appeared in Fritzing bins at the bottom of all the bins. In fact two searches appeared both with magnifiers but one has no entry field for searching. I also note that search.fzb reappeared in the OneDrive directory along with the my_search.fzb.

Hope this allows you to fix yours (even if a little clumsy solution) and the developers to hunt down a proper solution.

Alan

KjellMorgenstern, 2019-12-11:
A solution was proposed in #3576 to prevent / detect this kind of error

closed 2019-12-11 by KjellMorgenstern