fritzing/fritzing-app #3421: board-size differs when manufacturing

AlexanderEhle, 2018-09-03:
I upload my design to Aisler and get another board size than defined in Fritzing.
Examples: 28mm x 4.8mm is the Size in Fritzing. At Aisler you will get 27.79mm x 4.6mm.
This is 0.21mm and 0.2mm less than defined.
Also I upload a round pcb with 27mm diameter and get 26.95mm - 0.05mm.
Is there an internal rounding error?
[BoardSizeErrorSamples.zip](https://github.com/fritzing/fritzing-app/files/2345725/BoardSizeErrorSamples.zip)

KjellMorgenstern, 2019-07-15:
<tl;dr> 
I think the black line you see around the PCB is the area which is removed by the cutting tool.
So, the graphic in Fritzing is 28mm. Inside, you have a black line, forming a rectangle. The center of this line forms a 27.8mm wide rectangle, the inner of the black line is 27.6mm, the outer 28mm. That is what the cutting tool will remove.

The gerber contour output contains some deprecated commands, but seems legit, no rounding error.

`G04 MADE WITH FRITZING*
G04 WWW.FRITZING.ORG*
G04 SINGLE SIDED*
G04 HOLES NOT PLATED*
G04 CONTOUR ON CENTER OF CONTOUR VECTOR*
%ASAXBY*%
%FSLAX23Y23*%
%MOIN*%
%OFA0B0*%
%SFA1.0B1.0*%
%ADD10R,1.102360X0.188976*%
%ADD11C,0.008000*%
%ADD10C,0.008*%
%LNCONTOUR*%
G90*
G70*
G54D10*
G54D11*
X4Y185D02*
X1098Y185D01*
X1098Y4D01*
X4Y4D01*
X4Y185D01*
D02*
G04 End of contour*
M02*
`

In line %ADD10R,1.102360X0.188976*%  we can see the correct size (28mm in inch)
As I read it, inside this aperture, a rectangle is drawn, using D11, which is a 0.2mm circle.
Just guessing, that this line is then used for the cutting tool? Someone correct me on that.
So the rectangle for the cutting tool will be 27.79 x 4.6 mm, which is what is displayed after uploading to Aisler. 
The cutting tool might be 0.2mm width, and the result would be 27.6 x 4.4 mm.

KjellMorgenstern, 2019-07-15:
There is a black line around the PCB. Fritzing uses the outer of this line, while Aisler uses the center of this line.

failiz, 2021-03-12:
Duplicate of  #2993.

KjellMorgenstern, 2021-03-12:
Closing in favor of #2993

closed 2021-03-12 by KjellMorgenstern