fritzing/fritzing-app #3422: Parts update not working on Windows.

vanepp, 2018-09-13:
Parts update on Windows (all versions as far as I know, but I have only tried Win7) has stopped working. The problem looks to be the libgit2 version in the 9.3.b release only supports tls1.1 which it appears github no longer supports. The fix is to upgrate to at least libgit2 v0.26-3 (or later) which has a fix for "Enable Windows 7 and earlier to use TLS 1.2". Unfortunately this looks to need a recompile, I tied building a libgit2.dll from libgit2 v0.26-3 and copying just the dll in to the 9.3b directory, but parts update still doesn't work (I assume some other setting compiled in needs to change as well) although it does on a version of head compiled from source.

Peter

vanepp, 2018-09-21:
Is there someone working on this issue? I  have Qt5.6 downloaded and am downloading the source to the 9.3b release. I can compile a version of 9.3.b with a later version of libgit2 (and a later version of boost probably as well because I don't know what version was used in 9.3b, but I think that should all be internal to fritzing.exe and thus not an issue). That may produce a Fritzing.exe and new libgit2 dll that could be substituted in to an update of 9.3b to fix this issue. However if there is someone that actually has experience in building a release working on it it,  would be better to let them do it. An alternative may be to compile head to pick up the various translation improvements. Comments?

Peter

KjellMorgenstern, 2019-05-11:
PR #3467 might fix this.

vanepp, 2019-05-12:
No. #3467 won't, but a new release will. The problem is the version of libgit2 in 0.9.3b only supports tls1.1 on Win7 and github no longer supports tls1.1, compiling with a later version (only a couple of revs later) that supports tls1.2 on Win7 fixes it. I've known that for a long time as I could compile a version that works on Windows7 but I couldn't build a release version. You guys can and that will fix the problem. We can close this one as resolved (or about to be resolved) if you like.

KjellMorgenstern, 2019-05-12:
We use the latest libgit2 for upcoming 0.9.4

closed 2019-05-12 by KjellMorgenstern