fritzing/fritzing-app #3424: "Rotate part label 45° CCW" in German locale mislabeled

towe96, 2018-09-28:
When attempting to rotate a part label 45° counter-clockwise in the "circuit board" / "Leiterplatte" view, the option "Um 45° drehen (gegen den Uhrzeigersinn)" appears twice.
Selecting the first one actually rotates the label 135° counter-clockwise. The second instance works as expected.

vanepp, 2019-05-04:
In the english version we have 

rotate 45 clockwise
rotate 90 clockwise
rotate135  clockwise
rotate 180
rotate 135 counterclockwise
rotate 90 counterclockwise 
rotate 45 counterclockwise

and all the rotates work as expected. If I understand the problem here, the rotate135 counterclockwise is labeled "Um 45° drehen (gegen den Uhrzeigersinn)" instead of "Um 135° drehen (gegen den Uhrzeigersinn)" which would indicate the translate file for german needs a correction to 135 from 45 for the 5th entry. 

Looks like in 

fritzing-app/translations/fritzing_de.ts

line 5970 wants to change from 45 to 135 as the from translation is Rotate 135 Counter clockwise but the translation says 45 which is incorrect.

Peter

KjellMorgenstern, 2019-11-01:
Fixed by #3556

closed 2019-11-01 by KjellMorgenstern