fritzing/fritzing-app #3429: Generic IC is destroyed displayed in Schaltplan view

pinout65, 2018-12-04:
Hello,
[image]

after some changes in view Schaltplan and Leiterplatte the Generic IC is destroyed displayed in Schaltplan view. The 8pin IC-body is displayed at different place as the pin numbering ...

take care - fritzing is great ...

KjellMorgenstern, 2019-05-29:
This issue was closed due to missing information.
If you think this issue is still important, please provide additional information.
  - Which version of Fritzing have you been using? (0.9.3b?)
  - Which steps reproduce the problem?

KjellMorgenstern, 2019-05-29:
Note: I tried a few minutes some editing in schematic and pcb view with a generic ic, but did not stumble upon this issue. Did you use the autorouter?

closed 2019-05-29 by KjellMorgenstern