fritzing/fritzing-app #3430: Copper Fill at wrong place

pinout65, 2018-12-04:
Hello,

I want to Copper Fill the area
[image]

 next to F1 between the two small copper tracks and select it, but it fills next to U2 ... "Abstand der Massefüllung einstellen" is set to "0" ... IF set to "10" is filling on the right place, but it fills not all of the area ...

take care, fritzing is great ...

KjellMorgenstern, 2019-04-29:
You can help progress the issue by providing additional information.
- Attach your sketch file and/or custom part files to the bug report.

It would be great to have a minimal sketch to reproduce the problem.

KjellMorgenstern, 2019-10-12:
There is a significant distance between F1 and U2, so this clearly is a bug. But without files to reproduce it I can not help. If you can not publish the file, please contact me via PM.

pinout65, 2019-10-12:
Hello Kjell,

my versions of the project are attached ... I hope it help to find the 
bug ...

you do a great job - take care ...

Thies

KjellMorgenstern, 2019-10-12:
Github ate the email attachment. You can sent it to kjell@gmx.de or upload to this thread using a browser.

KjellMorgenstern, 2019-11-09:
Some improvements to copper fills and ground fills have been made. This includes a bug in the UI that would show the wrong GND seeds before a gound fills is made. I could not reproduce the exact issue described here, but it is likely fixed.

closed 2019-11-09 by KjellMorgenstern