fritzing/fritzing-app #3433: Diode Terminal numbering error in schematic

godwinhr, 2019-01-03:
When rotated the diode terminal numbering 1 and 2 remain on the same terminal.

HaraldRau, 2019-05-09:
Hello, can you load the Sketch here by filname.zip? I can not reproduce the problem in my environment!

godwinhr, 2019-05-18:
Hi, Please find attached the schematic.

Thanks,
Godwin

*"**If you have the desire and passion to do something, and it’s*

*within God’s will, you will achieve it**"*

KjellMorgenstern, 2019-07-15:
It seems attaching the file via email did not work?

godwinhr, 2019-07-21:
[Rectifier Schematic.zip](https://github.com/fritzing/fritzing-app/files/3414522/Rectifier.Schematic.zip)

KjellMorgenstern, 2019-10-12:
Using Fritzing g2e7bc9f5 , I rotated the Diode D1 multiple times. The result seems correct to me. Is this what you expected? 
[image]
[image]
[image]
[image]

KjellMorgenstern, 2019-10-22:
Closing as probably fixed / can not reproduce.

closed 2019-10-22 by KjellMorgenstern