fritzing/fritzing-app #3434: Fritzing unable to open components after last update

flamingomingo, 2019-01-05:
Fritzing (running on ubuntu 18.04) is not being able to open /core and show the message "Estamos tendo problemas com o mecanismo de mudança.
O Fritzing tentando resolver o problema, mas não será possível modificar as propriedas dos componentes."

After opening the components are blank.

vanepp, 2019-01-23:
You are better to ask questions like this in the Fritzing forums rather than here, that said  it sounds like you have a corrupted parts database. Here is a copy of the standard advise (you need to use the Linux paths rather than the windows ones at the bottom):

There are two user directories (with your parts and the parts database) which don't get touched during an install (to not affect your sketchs during upgrades). On Windows they are in

c:\users\username\AppData\Fritzing\roaming\Fritzing (which is a hidden directory so you need to enable hidden directories in explorer) and

c:\Users\username\My Documents\Fritzing (where username is your windows id)

If you don't have any parts or sketches you want to keep you can just delete those two directories and Fritzing will receate them, or you can move them aside by renaming them if you wan to keep something in them.

linux

~/Documents/Fritzing/parts
~/.config/Fritzing 

Peter

flamingomingo, 2019-01-25:
ok thanks.. I solved the problem.

Just to clarify I asked here in git because the fritzing software directs me directly to this page. But it will not be repeated.

Thanks againd...

vanepp, 2019-01-26:
Good to hear it fixed your problem. This wasn't meant as criticism, just advise. Development of Friting has pretty much died off, (I'm trying to restart it, so far with limited success) which means that the few of us that are still interested (and therefore able to help) are mostly in the Fritzing forums rather than in here so you have a better chance of getting a reply in the forums. 

Peter

KjellMorgenstern, 2019-05-11:
PR #3467 might fix this.

closed 2019-12-11 by KjellMorgenstern