fritzing/fritzing-app #3437: Parts Update Results In "Unable to open temporary file" #3308

beetlegigg, 2019-02-13:
The issue described in issue #3308 is marked as closed.   But today 13th Feb 2019 I have exactly this issue with version 0.9.3 running on my mac os 10.14.1

I cannot update the parts database with the latest and part regenerate reports cannot open temporary file.  Hoping there is a fix this. Thanks

beetlegigg, 2019-02-14:
Oh.. now I see from reading some other issues that Fritzing is not being maintained and is limping along to its demise.  What a b*gger.  I only just started with it a few months ago.  I suppose that means I've go to try to get to grips with the likes of KiCad and that looks bloody complicated.  Its all doom and gloom.

beetlegigg, 2019-02-14:
I should still leave this as open I suppose.

vanepp, 2019-02-16:
Hope isn't entirely lost, some of us are trying to restart development (see issue 3435 if you haven't). It probably won't be short term (I have been trying without much success for the last two years) but there is more hope than there has been for a while. 

Peter

KjellMorgenstern, 2019-05-11:
PR #3467 might fix this.

closed 2019-12-11 by KjellMorgenstern