fritzing/fritzing-app #3439: local file storage

CarminKern, 2019-03-08:
How can I Change local file storage for fritzing?

HaraldRau, 2019-03-08:
Do you mean the Storage for the parts? ../fritzing/fritzing-parts/

Which Storage do you want to include? Your local store in C\: or a Git-Repository?

CarminKern, 2019-03-08:
I mean 
 C:\Documents and Settings\[user name]\Application Data\Fritzing\.

HaraldRau, 2019-03-09:
Sorry, I do not know, how to change the part storage.

vanepp, 2019-04-27:
Depends on how you want to change it. The path is hard coded in to the source, so you need to compile from source to make name changes. You can change where it is stored by using symlinks to change which disk the directories will be written to (I used to do that to get them of the Windows system disk) but that is about all without source level changes.

Peter

KjellMorgenstern, 2019-05-04:
Fritzing part folder structure is discussed in #3454

closed 2019-05-04 by KjellMorgenstern