fritzing/fritzing-app #3441: database regen relies on folder and file with INVALID file names in windows.

Feay, 2019-04-21:
database regen and reading relies on folder and file with INVALID file names in windows. Please linux people stop punishing us!
 DOTwhatever  is an invalid file name and folder name in windows, the user cannot create it, and if they change it cannot ever change it back, however for some (good) reason  7zip is able to unzip and create these invalid folders and files spacficly ..\fritzing-parts\.git and ..\fritzing-parts\.gitignore please note the DOT!
IF the user renames these, deletes them, or simply replaces their fritzing-parts folder with the https://github.com/fritzing/fritzing-parts/ zip download to attempt to update...
FRITZING WILL FAIL TO CREATE DATABASE and then FAIL TO SWAP  error.
this needs to be fixed and added to documentation. More over https://github.com/fritzing/fritzing-parts/ does not contain this folder and file that are necessary for fritzing to even work. I spent 4 hours messing around to get updated parts to finally figure this out. No help was available on your website or forums, just mystery.

KjellMorgenstern, 2019-04-21:
Can not reproduce. Note: Please list steps to reproduce the issue, or include screenshots of error messages, otherwise you are unlikely to receive help. Also, please don't YELL.

closed 2019-04-21 by KjellMorgenstern

Feay, 2019-04-29:
it's simple if anyone just replaces their parts folder with the new parts
folder IE not including or deletes the hidden folders that are invalid
names in windows ie .git then the program will mess up. this is an issue
and or a feature request. more over these invalidly named folders hold no
parts information! they just consume twice the amount of space for no good
reason. I know i know you're probly a linux person that hates everyone
else, but please understand you cannot name a folder with a DOT as the
first character in windows. the ONLY way they get created is they are
prepackaged in a zip file and your flavor of unzip program is smart enough
to do whatever.
and there for this is another issue bad documentation! atleast tell people
how to properly update their parts folder with out messing it all up. IF
you wanna make a thing atleast it ought do what it's ment too and it be
made know how it works.

KjellMorgenstern, 2019-04-29:
This is more about patience than about hate. If you want to help, please refer to the specific passage in our documentation about the parts that should be improved.

About the .dot files:
> However, it is acceptable to specify a period as the first character of a name. For example, ".temp".
https://docs.microsoft.com/en-us/windows/desktop/fileio/naming-a-file