fritzing/fritzing-app #3443: Fritzing 2019 Roadmap

KjellMorgenstern, 2019-04-27:
Hello,

as announced in #3435 , I am taking over maintenance of Fritzing for André. My C++ / Embedded development / Robotics background is a good start, but I am not alone. Together with the help of the people at Aisler and @ovidiub13, a developer with KDE background, we plan to release a fresh build of Fritzing in the near future. 

So, this is a short term roadmap, limited to the minimal things to get things rolling.

**April/May 2019**
1. Add CI for Linux, Windows and Mac targets
2. We will create a test release 0.9.4alpha , since toolchains and many other things have changed, and we need to verify that processes still work on a all platforms.
3. Reduce the number of open issues, by closing issues which are too old, no longer valid, or just to low in priority as “No fix, please re-open if you disagree”. 
4. After alpha, we will roll out 0.9.4b to the download section.
5. Update developer wiki: Align better with git-flow, rebase all pull requests before merge, only accept PR if CI is green, check basic coding style...

**June/July 2019**
8. (ongoing) Reduce the number of open issues...
9. Prepare 0.9.5 release, work on new features/feature requests
10. Add black box and unit test infrastructure

 **August/September 2019**
13. Create 0.9.5 beta
14. Release 0.9.5 stable

vanepp, 2019-04-27:
This is wonderful news,! I have been trying (with little success) for that last couple of years to restart development.

"We will create a test release 0.9.4alpha , since toolchains and many other things have changed, and we need to verify that processes still work on a all platforms."

This shouldn't be to bad. I have working environments for Linux and Windows7 and 10 with current versions of Qt and the dependencies. Libgit2 on Windows was hard (at least for me) but I now have a working version (I can share what worked for me, if it turns out to be not obvious for someone experienced in development). There was a problem with an assert on tspans in Qt but I fixed the parts that were causing it in the parts database and it now builds correctly (or at least did a couple of months ago) on Linux and Windows. I don't have a Mac, and someone that does tried but failed (libgit2 problems) to get a build going there, but I expect that to be only a problem of configuration (Linux is somewhat picky about configuration too). Lately I have been working on updating the parts templates on the web site as the current ones are just plain wrong. 

Peter

karpawich, 2019-04-27:
I'm so excited that AISLER found a maintainer! That is great news. @KjellMorgenstern I've been doing some work to rebuild Fritzing in JavaScript. Seeing as you've been brought in, I'll transition towards building a Parts Editor instead.

>     5. Update developer wiki: Align better with git-flow, rebase all pull requests before merge, only accept PR if CI is green, check basic coding style...

When I wrote parsing functions for Fritzing data files, I discovered many components which were left undocumented in the Wiki. Once when you shift focus to the Wiki, I'll open an Issue with the missing components so the new Wiki can be complete.

jchull, 2019-04-30:
> ... I've been doing some work to rebuild Fritzing in JavaScript.
Is there a public repo? Seems like someone interpreted your comment to mean the app is being rewritten in JS.

monodesigns, 2019-04-30:
@jchull 
> Seems like someone interpreted your comment to mean the app is being rewritten in JS.

I got here from a [Hackaday article](https://hackaday.com/2019/04/30/fritzing-is-back-and-this-time-its-written-in-javascript/) that claims that Fritzing is being rewritten in JS.

deshipu, 2019-04-30:
@jchull The effort to "rewrite" Fritzing is centered at https://github.com/freetzing

KjellMorgenstern, 2019-04-30:
@monodesigns Thanks for the link. The article is misleading, the releases planned here are based on the current code (that is in C++, just to be clear).

karpawich, 2019-05-01:
@deshipu @KjellMorgenstern @jchull @monodesigns 

I originally led the team which was porting Fritzing over to JavaScript for future development. Seeing as Kjell has been brought in to continue development on the existing C++/Qt codebase, I will be transitioning my work with JavaScript to a Parts Editor desktop application.

I will contact the Hackaday team as well to see that the article is updated to reflect the recent changes.

KingDarBoja, 2019-05-03:
That's a great news! Been waiting for some clean up on Fritzing repo and some development since this open source software been around. 

Also, there is going to be some work on the official website along with the forums after those releases? Just curious.

Cheers!

ovidiub13, 2019-05-03:
Hi @KingDarBoja. Yes we plan to work on the website as well. We're still in the planning phase on that, as there are many moving parts that involve the site.

DonPavlov, 2019-05-08:
Any updates on the toolchain and the setup guide for compiling it? I currently fail because of libgit2.

KjellMorgenstern, 2019-05-08:
@DonPavlov You can use the develop branch. No known issues with libgit2, it stills needs to be next 
 to the fritzing-app directory as described in the wiki. Currently we use 0.28.1, but older versions should also be ok. You can also get some ideas from the Dockerfile or the .travis.yml or appveyor.yml if you are stuck, those files contain all the stuff for automated builds.

DonPavlov, 2019-05-09:
@KjellMorgenstern will try out again with 0.28.1 was just following the guide, which states to use 0.23.4 and somehow i ran into an error what was one of the older issues. It did not find the included lib. Will take a better look at it again with the new version.

IvanDelsinne, 2019-05-20:
With a background in both backend development and sys administration, I guess I could provide some help with the website's hosting

KjellMorgenstern, 2019-05-20:
The website is currently in a private repo, which is of course bad for colaboration. This was all for good reasons (trying to give Fritzing to some reliable constant financing), but it didn't work out as planned.
From the technical side, the webpage is written in python, and about 10 years old. Short term plan for now is to (re)build static parts of the page.

vanepp, 2019-05-20:
@KjellMorgenstern The main site web server looks to be having performance issues at the moment. It has always been slow to log in to the forums, but in the last week or so it has started to sometimes time out before responding with the login page. The forums (which I gather are on a different web server) seem to be fine.

KjellMorgenstern, 2019-05-20:
Yes, the forum is fine more or less, with an up to date stack. However the registration part is still on the old server.

KjellMorgenstern, 2019-06-20:
Update on the roadmap: May has passed, and we don't have an official 0.9.4 release, so we are behind the roadmap I set up two months ago.
The reasons:
1. A lot of work is going into getting fritzing.org more reliable. This includes some fixes on the current server, but also a complete rewrite of the site.
2. Also, we decided to establish a full continuous delivery system first, which is about to go live: PR #3488 . That means for each new fix a runnable developer version of Fritzing can be downloaded from github.

vanepp, 2019-06-20:
For me (and I expect most everybody else too) I'm happy to see progress (and there has been a lot!) A suggestion for the dev builds (if possible), I modify 

src/utils/folderutils.cpp like this:

at line 144 change

    QString path = QSettings(QSettings::IniFormat,QSettings::UserScope,"Fritzing_dev","Fritzing").fileName();

to

    QString path = QSettings(QSettings::IniFormat,QSettings::UserScope,"Fritzing_dev","Fritzing_dev").fileName();

and at line 151

    return dir.absoluteFilePath("Fritzing");

to

    return dir.absoluteFilePath("Fritzing_dev");

This means the the dev build will use different user directories than 0.9.3b so both can coexist at the same time on the same system without affecting the other. When loading a dev version you can either use the Fritzing_dev directory as is (Fritzing will create it if it doesn't exist) or copy the Fritzing directory from 0.9.3b in to Fritzing_dev to make the dev version identical to your production version. I find this makes testing new versions much easier.

Peter

wraybowling, 2019-09-23:
I'm very happy to see this. I truly believe in Fritzing. I want to contribute some of my deeper understanding of SVGs and Inkscape as well as user experience design, so I'll see about getting involved behind you in those ways. Looking forward to it.

KjellMorgenstern, 2019-09-25:
@wraybowling SVGs and Inkscape, that means the part editor, which is great as there are certainly some improvements possible. You might want to look at the issues with the labels https://github.com/fritzing/fritzing-app/labels/Parts-Editor and https://github.com/fritzing/fritzing-app/labels/Parts to see if there is something for you.

vanepp, 2019-09-25:
@wraybowling If you don't already know of it, the parts file format document is very useful for understanding the parts files xml (unfortunately not all of it though, and I am not sure it is completely up to date with the current code) in the blog at https://github.com/fritzing/fritzing-app/wiki/2.1-Part-file-format is an excellent resource. Someone familiar with Inkscape's quirks will also be very welcome. I understand very little of it, and it tends to do odd things (probably because I don't understand what I have done). You can find discussion on a number of the issues in the Fritzing forums. The parts editor was not finished when development stopped, so I tend to edit the underlying files and ignore parts editor. Making parts creation easier in any manner (Suggested Inkscape preference file settings, svg tips, almost anything) would be welcome. It currently has a steep learning curve. 

Peter

ju5t, 2020-05-30:
Any news on the Catalina builds?

On the website it said 10.15 is untested but it's actually unsupported because it simply doesn't start. It was only after the payment it said the download was there for 10.14.

I don't mind the money, but you could do a little better on the information side of things to be fair.

KjellMorgenstern, 2020-05-30:
@ju5t I tried it on a VM with a fresh MacOS Catalina, and it started flawlessly. But that is not enough to claim it is tested or supported. MacOS Catalina is from October, to late to be considered for the 0.9.4 release of Fritzing in November. The most common issue is that permission to start must be set first, as MacOS won't allow it to run without an official certificate.

ArsineMan, 2020-06-09:
Hi. I am having a hard time understanding where Fritzing lives.  I see fritzing.org has a 9.4 version, I see someone now has a freetzing and I see this repository.   Can anyone direct me to the most widely used codebase for fritzing?

KjellMorgenstern, 2020-06-09:
@ArsineMan This one... compare also to fritzing.org , or compare the github activities of other projects.

KingDarBoja, 2020-06-09:
@ArsineMan ignore the Freetzing repos as that was a previous try from community to migrate Fritzing into JS/TS environment, but soon after that, Fritzing got new maintainers like Kjell and now the projects is alive again.

This repo is the official and only source code of Fritzing 🚀

Janhgm, 2020-07-21:
@KjellMorgenstern 

Is development for Fritzing possible for others. 
Like to know if github can be used or that current development is in isolation.

KjellMorgenstern, 2020-07-29:
@Janhgm We accept pull-requests, although we might be picky. Try to keep them small for easy and fast review, add a description, and link the corresponding issue on github. The onboarding is quite time intensive if you don't yet have a lot of experience with C++ and Qt framework. If you are not, I recommend starters to get familiar with the parts format and the fritzing-parts repo. Working on parts requires some basic understanding of electronics, and SVG file format, and Fritzing specific XML. Python is also helpful. Improving parts availability is also an area were we want to focus on in the near future.

Janhgm, 2020-07-29:
@KjellMorgenstern .  OK.  I shall start with an isolated copy to make my own build and to look how well the code is maintainable. (Got some experience with other open projects where I was admin.)

RamblingGeekUK, 2021-01-17:
How is this going  ?

failiz, 2021-01-19:
It is going well. Kjell is working hard and he will release 0.9.5 soon. However, please, do not ask these type of questions in the issue tracker, ask them in the forum. It wastes developer time (we get emails for each new comment in the issue tracker). See for example: 
https://medium.com/@methane/why-you-must-not-ask-questions-on-github-issues-51d741d83fde

failiz, 2021-04-25:
Time to make a roadmap for 0.9.7!
Probably the best is to close this issue and open a new one.
Regarding me, I will continue working on the simulator: 

- Add a few more spice fields and fix some of the current ones
- Create examples
- Show spice properties in the inspector
- Maybe trigger simulation when parts properties change?
- If I have time, make potentiometers and buttons interactive (be able to change their properties without using the inspector)
- Fix bugs, clean the code (remove WX code) and merge it into develop

My two cents, it would be nice if we could address any of the main problems:

- SVG compatibility 
- Parts editor
- Gerber export issues
- Automatic checks for new parts and easy way to submit them
- Properties handling (maybe not the most urgent)

th-gauweiler, 2021-04-25:
Fine.
How about all errors which are marked with high priority?
E.g.  #2199 was found nearly 10 years ago.
(yes I know it was long time ago the current team was etablished.)

Just my two cent.

Thomas