fritzing/fritzing-app #3452: declaration different from implementation

maddin200, 2019-05-02:
/src/utils/folderutils.cpp: line 657 

static void copyBin(const QString & source, const QString & dest);
vs
void FolderUtils::copyBin(const QString & dest, const QString & source) {

vanepp, 2019-05-04:
For what it is worth (which may not be much :-) ), here is how I would go about resolving, this assuming you can't find someone familiar with the code to answer (which I never have):

vanepp@vanepp-All-Series:~/fritzing-app$ grep -r copyBin .
./src/mainwindow/mainwindow.cpp:    FolderUtils::copyBin(binFileName, BinManager::TempPartsBinTemplateLocation);
./src/fapplication.cpp:    FolderUtils::copyBin(BinManager::MyPartsBinLocation, BinManager::MyPartsBinTemplateLocation);
./src/fapplication.cpp:    FolderUtils::copyBin(BinManager::SearchBinLocation, BinManager::SearchBinTemplateLocation);
./src/utils/folderutils.cpp:void FolderUtils::copyBin(const QString & dest, const QString & source) {
./src/utils/folderutils.h:    static void copyBin(const QString & source, const QString & dest);

That tells me where this is called. so I can try and figure out what will cause mainwindow.cpp to call this code (presumably loading a bin) and create a test case in Fritzing to load the bin. Now gdb allows me to trace the code to figure out what should be happening. I think the likely thing is that the .h file dest and source should be reversed and this is just a typo, but that could be wrong if it is in fact copying the other direction (i.e from BinManager::TempPartsBinTemplateLocation to binFileName instead of the other way around.) This is time consuming, but the only way I have found to figure out the code in the absence of either documentation (which I'm not aware of any) or someone familiar with the code (also not aware of anyone.) Usually this will cause a search to try and figure out what BinManager::TempPartsBinTemplateLocation is and does. It may be either where the bin will be stored or a template file that the bin will be created from. It would be a valuable thing to document what we find (I try and keep notes when I do a search like this) to begin documenting the code, since that is sorely needed.

Peter

KjellMorgenstern, 2019-05-04:
Changing the header will keep the current behaviour, which is the safe bet. Looking at the implementation it is quite clear that the header file is wrong: The method silently returns if destination exists (which is somewhat troublesome by itself) 

But of course it make sense to check if someone tripped over that. The occurences you found use a template as source, which seems legit.

closed 2019-10-09 by KjellMorgenstern