fritzing/fritzing-app #3453: Use BOOST_ROOT

KjellMorgenstern, 2019-05-03:
I propose the following behavior:

Remove the usage of relative or absolute directories. 
Always try BOOST_ROOT first, and if that is not set, try system boost. Fail otherwise.

vanepp, 2019-05-04:
That seems sensible to me. Although I wasn't involved in Fritzing when these decisions were made, I think the idea was to make Fritzing self sufficient to make the install simple and relatively foolproof (and that has worked fairly well). As long as we provide an explanation of what is missing to the user if boost isn't found so it remains at least reasonably easy to get working that should be fine. 

Peter

KjellMorgenstern, 2019-05-18:
Well, you need boost to compile Fritzing, there is no way around that. Having to copy it to some magic location that is even *inside* the fritzing directory is not ideal.

KjellMorgenstern, 2021-07-06:
Current way of having boost next to the fritzing-app directory is working fine.

closed 2021-07-06 by KjellMorgenstern