fritzing/fritzing-app #3454: Sort out part folder locations

KjellMorgenstern, 2019-05-03:
There are quite a number of issues and fixes to where the part folders should be located, and what about it causes problems. Before any fixes are accepted, or even further code is written, I'd like to collect and discuss the locations here. As a start, I propose.

Parts are always stored in a hidden user directory, like ${HOME}/.local on linux.

gasinvein, 2019-05-03:
Yeah, parts manages by Fritzing itself (e.g. for updates from GUI) should be placed in writable, per-user directory, e.g. `${XDG_DATA_HOME}/Fritzing/parts`.
See flathub/org.fritzing.Fritzing#1 for example of a downstream issue.

KjellMorgenstern, 2019-05-03:
Open questions
- Where to store on Windows?
- Where to store on MacOS?
- How can the user import fzbz files?
- How to protect the user from directly messing with ${PARTS_DIR} ?

gasinvein, 2019-05-03:
@KjellMorgenstern 
>* Where to store on Windows?
>* Where to store on MacOS?

I'm not familliar with Qt, but AFAIK it has `QStandardPaths::AppDataLocation` for that.

cjmayo, 2019-05-03:
AppDataLocation will also give the fallback to /usr:

```
AppDataLocation "~/.local/share/<APPNAME>", "/usr/local/share/<APPNAME>", "/usr/share/<APPNAME>"
```
https://doc.qt.io/qt-5/qstandardpaths.html

I use the home directory and don't install in /usr but I think it would be good to support:
1) Users who don't want to or can't rely on the Internet
2) Shared computers where admins don't want multiple copies of the library (and maybe prevent user updates).

Education (I'm guessing, not from experience) would be a use case for these, and an important potential user base for Fritzing.

But AppDataLocation, no mandatory dependency on Git, only enabling updates if library is writeable and making snapshots of the library available with every app release should cover this?

gasinvein, 2019-05-03:
>AppDataLocation, no mandatory dependency on Git, only enabling updates if library is writeable and making snapshots of the library available with every app release should cover this?

I _think_ it would be the optimal approach for Linux. Can't say for other platforms.

vanepp, 2019-05-04:
A couple more feature requests from the Fritzing forums as well: Someone in the last few days wanted to import the adafruit Fritzing repo in to core parts because he mounts the parts repo readonly and and NFS exports it to client machines in a library (with presumably 100s to 1000s of different users). He doesn't want to have the repo replicated in every client's local directory as that wastes disk space, only on the read only server (which means it needs to be in core parts which conflicts with the automatic update mechanism). While I figured out how he could do that with some manual work for the adafruit repo, he doesn't seem interested, but the same request has come from educators that want to do the same type of thing: have a centrally maintained parts repository that they can export read only to many client machines over the network. Another request of a similar nature was for a flag to turn off the parts update check on the repo to save the time that takes when the parts repo is read only, as the check time (which can be a couple of minutes and probably significant network load on a class room full of machines all starting at once.) I think the use of .gitignore may allow someone to add repos such as Adafruit's to an install while still allowing the automatic update from github to work, but I don't know enough about git to know for sure.  As well there have been requests to make the user directories more friendly to run on USB Keys, which require being able to either set or automatically detect a different and changing from machine to machine path to the directories. There is apparently some standard way to do that on at least Windows, but I don't know what it is. If we can figure out some way to do these things there are people who would like to see them. 

Peter

KjellMorgenstern, 2019-05-04:
@gasinvein Thanks a lot for the hint, taken in.

@cjmayo I think your concerns are covered: 1) Parts are still packed with the release, and installed to ${PARTS_DIR}. Updates are fetched from github by default, but could be configured for an offline source if wanted. 2) I don't think protecting a (public?) workstation from user modifications is in the scope of fritzing, there are systems for that. What we could do is download the parts if the parts directory was deleted.

@vanepp How is disk space an issue? It is measured in GB, and the parts directory is well below that. In a setup were disk space is that spare, network bandwidth will probably not be vast as well. For the other concerns (distribution, usb disk) do you think setup descirbed above, with a fixed parts folder and a configurable update source would work?

gasinvein, 2019-05-04:
@KjellMorgenstern Could Fritzing potentionally support multiple locations? E.g. first look in per-user directory, then in system-wide directory.

KjellMorgenstern, 2019-05-04:
@gasinvent  Do you mean multiple at the same time, or using the system dir if no parts directory exists in the user dir?

gasinvein, 2019-05-04:
@KjellMorgenstern At the same time, with user's directory taking 
preference.

KjellMorgenstern, 2019-05-05:
@gasinvein Could you elaborate on the use case for this? If a user parts directory is present, at what time should the system directory be accessed?

gasinvein, 2019-05-05:
@KjellMorgenstern Well, I meant loading parts from both directories, e.g. load everything from per-user directory and then load from system-wide directory only those parts not present in per-user directory.

KjellMorgenstern, 2019-05-05:
@gasinvein There is the existing mechanism to index both, parts from the app and parts from the user (from QStandardPaths::DocumentsLocation /Fritzing/parts) . All is read into an sqlite database. How those parts are merged is a bit beyond the scope of this issue, but basically the app parts are read first, and the user parts second, so they should overwrite the ones provided by the app. The code for this is in PaletteModel.cpp, loadParts.

In the long term, it think it is better to encourage the use of document management (in our case, loading parts from a git repo), instead of throwing around zip files. This is to reduce the amount of defective or outdated parts, or at least to be able to warn users that a part is EOL. In general, it improves collaboration (imagine an open source project that relies on .zip files for working on the source code. Possible, but...)

deshipu, 2019-05-05:
> (imagine an open source project that relies on .zip files for working on the source code. Possible, but...)

Actually, if I make a design for a project, I want all the part files and other relevant dependencies to be included in the project, in exactly the version I used in my design. Otherwise using an older project, when there have been a lot of changes to the parts, would require a lot of work fixing all the differences first.

KjellMorgenstern, 2019-05-05:
@deshipu I am not sure if you meant that as a point against or for using a version control system?

deshipu, 2019-05-05:
@KjellMorgenstern kinda. I mean it as a point against using a single, globally configured, auto-updating, hard-coded repository, which seems to be the current preference.

KjellMorgenstern, 2019-05-05:
I get it. The new idea is to make the source configurable, so the user, or the admin, would be able to configure which git repo and branch should be used to supply the parts. That would give the user full control.

vanepp, 2019-05-05:
@KjellMorgenstern "How is disk space an issue? " with the warning that I'm speculating from the description of his installation so may not be entirely correct, he has a server that contains the Fritzing app and the core parts data base. This is NFS exported readonly to I think around 8 client machines that the public can use. The location is a public library which potentially has 1000s, to 10s of thousands of users on those 8 machines in a timeshare environment. The disk space issue is with those user's home directories. His particular desire is to allow the Adafruit parts repo (which isn't in core parts and has a fair number of parts) to be accessed by the users. Loading the bin (as would normally be done) will make a copy for each user that uses it in their account's local disk space. The number of users in a time share environment (as opposed to our usual model of one user per machine) multiplies the disk space aspect in unusual ways. I use to work for a university data center with a 40,000 user time share system so I'm familiar with the issue. What he wanted to do is load it in to core parts to have only one copy shared over potentially thousands of users which is possible with a bit of work to convert the bins to internal core format. He didn't seem interested in doing that work so the issue has died. Schools face a similar problem, there is desire to be able to manage a single readonly parts repo (both for ease of maintenance and update and to keep students from hacking the parts for other users) with the same issue, with per user storage of extensions, disk space multiplies by the number of users. If we can find a way to support that easily it would be useful to at least some of the user base.

deshipu, 2019-05-05:
@KjellMorgenstern that doesn't really solve the "old project using old parts" problem, since you would need to keep a repository of old parts and to change the path to it in settings every time you open an old project. A per-project repository could work, but even better would be if any parts stored in the same directory as the project's file had priority over any other parts — then you can simply commit your parts with the rest of your project, and update them explicitly when needed, without having to worry about newer versions of Fritzing or non-standard configurations.

vanepp, 2019-05-06:
@deshipu  This issue is covered by the parts obsoleting mechanism. When a part is replaced in the parts repo the old part remains in the obsolete section with its original moduleId and a new moduleId is assigned to the updated part (and the pins are mapped to the new part if required). When you load an old sketch it will find the parts in the obsolete parts bin and offer to replace them with the new parts if you wish. As far as I know this is not documented (although as with many things, it should be) and is done by the parts maintainer as part of QA for new parts.

deshipu, 2019-05-06:
I see, thank you.

KjellMorgenstern, 2019-07-09:
Commenting on https://github.com/fritzing/fritzing-app/issues/3443#issuecomment-504044033 from @vanepp 

The continuous build version should deviate as less as possible from the released versions. But how about reading the Fritzing directory location from the config file? And of course, the name/location of the config file would be made a command line parameter.

I think that would help in this case, and also be handy in other use cases?

vanepp, 2019-07-10:
Being able to change the location via a command line parameter should fix both the dev version issue and the USB key issue (at least I think so. as I'm not entirely sure of the issue on the USB key side, but I think they want all data on the USB key and this should allow that.)

Peter

michaellass, 2019-11-12:
I'm trying to package Fritzing 0.9.4b for Arch Linux and I'm still struggling with this whole topic.

Fritzing tries to find core.fzb and contribParts.fzb under $HOME/bins/ and obviously fails. Parts in /usr/share/fritzing/parts are not considered at all. Even specifying the location with the "-f" argument makes no difference whatsoever.

How is this currently supposed to work? And what do I need to do to get it working? Some compile time flag? Some runtime arguments?

For 0.9.3b we used the patches from #3238. However, they don't apply to 0.9.4b. Any help is appreciated. Or should we just wait for 0.9.5?

KjellMorgenstern, 2019-11-13:
@michaellass The known issue with installing Fritzing Parts at a write protected location have not yet been fixed: Fritzing wants to update the parts, but can not write, will fail and give a not so helpful error message, but should continue to work.

Basically, installing fritzing to /usr/share/fritzing should work, with said limitation of write protection. I just tried it on a fresh minimal Ubuntu 19.10 installation. After installing some Qt packages, everything seems to work fine. I'll investigate a bit more.

michaellass, 2019-11-13:
Thanks @KjellMorgenstern, I think this now gives me an idea why me build does not work. I get a totally different file structure out of it:

I'm just doing the following:
```
tar xf CD-415.tar.gz
cd fritzing-app-CD-415
sed -i 's/LIBGIT_STATIC = true/LIBGIT_STATIC = false/' phoenix.pro
qmake-qt5
make
mkdir target
make INSTALL_ROOT=target install
```

Now looking into the target directory, I get the following structure:

No wrapper script anywhere
Executable: /usr/bin/Fritzing
Content of /usr/share/fritzing: help, sketches, translations

If I extract fritzing-parts into a subfolder called "parts" before running qmake and make, /usr/share/fritzing will additionally contain the parts directory. So I assumed this is the location where it has to be.

Extracting the parts into a subfolder "fritzing-parts" will _not_ cause make install to include this directory in the installation.

So I guess the way to go is building without the parts and the moving them to the right place manually. I'll try to go forward this way.

KjellMorgenstern, 2019-11-13:
Try
```
cd fritzing-app-CD-415
./tools/linux_release_script/release.sh 0.9.4-arch
```

cjmayo, 2019-11-13:
> For 0.9.3b we used the patches from #3238. However, they don't apply to 0.9.4b. Any help is appreciated. Or should we just wait for 0.9.5?

FWIW I've just rebased #3238 on CD-415. I haven't used it much but it does compile (against develop) and parts show up in the interface!

michaellass, 2019-11-13:
Thanks to both of you! Looking at the release.sh script, I was able to build a working package now. I'll try to include the patch by @cjmayo again so that users can create a local copy of the parts db if they want to.

KjellMorgenstern, 2019-11-13:
Please consider this very carfully, as it is often impossible to tell (for us and for users) what the correct part folder location is.

KjellMorgenstern, 2019-11-13:
So in the short term, I think it is better to disable the updates, and just use the package as is. In the mid term three things will solve this:
1. More frequent releases 
2. Configurable parts folder
3. Not clean, but maybe necessary: Help/debug dialog in "Preferences" that shows which folders are used (including write protection)

KjellMorgenstern, 2019-11-24:
I investigated a bit, so here is an update on this.

I am using a custom assets folder (for parts repository and sqlite database) in my development environment all the time, this requires no code changes (probably works back to 0.9.3)

Probably most uses cases can be covered by calling Fritzing once with the correct combination of 
--database --parts and --folder parameters. 

For using Fritzing in a users only folder, for example set ${release_folder} to $HOME/fritzing-0.9.4 and run
```
git clone https://github.com/fritzing/fritzing-parts $HOME/fritzing-0.9.4/fritzing-parts
Fritzing --parts "${release_folder}/fritzing-parts"

```
After this is used once, locations are stored in the Fritzing.conf file, like
```
0\location=user
0\path=/home/user/Documents/Fritzing/bins/search.fzb
0\title=Search
1\location=app
1\path=/home/user/fritzing-0.9.4/fritzing-parts/bins/core.fzb
1\title=Core Parts
```
Subsequent executions of Fritzing will stick with values from this config. 
Parts repo can be changed by again calling Fritzing with the --parts option.

akkana, 2020-02-27:
> probably works back to 0.9.3

Nope -- at least, 0.9.3b on Ubuntu 19.10 ignores the config file. Editing ~/.config/Fritzing/Fritzing.conf to change /home/akkana/Documents/Fritzing to another directory in the two places where it occurs doesn't stop Fritzing from creating ~/Documents every time it runs, and it doesn't see user-installed parts. Running with -f path-to-config-dir doesn't help either.

Can't report on 0.9.4; still struggling with the build system.

KjellMorgenstern, 2020-02-27:
@akkana I thing you are talking about the user folder (where user parts and projects are stored), not the parts folder. To be clear: The main issues is that the parts folder comes with the distribution, but Fritzing expects it at a user writable location. A build of 0.9.4 for ubuntu is available and widely used.

akkana, 2020-02-27:
@KjellMorgenstern Yes, you're right, I'm talking about the user folder. I want to put it somewhere other than ~/Documents.

I tried this:
- rm -rf ~/.config/Fritzing
- rm -rf ~/Documents
- download, un-tar and cd into fritzing-[longid]-master-498.bionic.linux.AMD64
- ./Fritzing (let it start up, then exited without doing anything) -- now I know I have a brand new config that's only from 0.9.4, nothing remaining from 0.9.3
- Edit ~/.config/Fritzing/Fritzing.conf and replace /home/akkana/Documents/Fritzing with /home/akkana/Fritzing in the two places where Documents occurs
- rm -rf ~/Documents
- Run Fritzing again, ./Fritzing, and exit without doing anything

Now ~/.config/Fritzing/Fritzing.conf has my edits replaced, with Documents back in the two places where I removed it, and ~/Documents/ has been created again. 0.9.4, like 0.9.3, ignored my changes to the Fritzing.conf file.

BTW, I admit that I haven't paid for this download, yet. But I promise I will if I can find a way around this regression.

KjellMorgenstern, 2020-02-27:
@akkana This is issue #3350, it will not be implemented in the near future.

vanepp, 2020-02-28:
@akkana AFAIK this is hard codedat present  (at least in 0.9.3b, I haven't yet looked at 0.94). If you like I can dig up the change I make on my dev system so the user files are different than the prod version for testing purposes (its a couple of line source change), but it means building from source which is a lot of work. I didn't see anything that overrides the value from a  config file (only in 0.9.3b though) although I'm new to the code and may well have missed something.

akkana, 2020-02-29:
@vanepp I found a workaround: there doesn't seem to be an environment variable for the documents directory, but you can set HOME before running Fritzing and then it will create Documents under whatever directory you set that to.

KjellMorgenstern, 2021-07-13:
# Summary

## Parts
From 0.9.7 on, Fritzing will *not* automatically try to update the parts.

Updating or changing the parts repo manually (of course with matching permissions for the users individual setup) via git command line or any other tool is still possible, and the implementation stays in Fritzing for further improvements.

## Documents
The folder for user files and parts follows the standard specified by their OS. These are explained in detail in https://doc.qt.io/qt-5/qstandardpaths.html  . Namely for Documents, this is

Path type | macOS | Windows | Linux
-- | -- | -- | --
DocumentsLocation | "\~/Documents" | "C:/Users/\<USER\>/Documents" | "\~/Documents"
AppConfigLocation | "\~/Library/Preferences/\<APPNAME\>" | "C:/Users/\<USER\>/AppData/Local/\<APPNAME\>", "C:/ProgramData/\<APPNAME\>" | "\~/.config/\<APPNAME\>", "/etc/xdg/\<APPNAME\>"

## Config
Settings are stored in the Registry (Windows), and in a Frtizing.conf file in AppConfigLocation (Linux)

## Customization
The --parts parameter can be used to set a specific path for Fritzing parts. It needs to be passed each time Fritzing is started.

KjellMorgenstern, 2021-07-13:
PS. 0.9.7 introduces a feature that fzp files can be imported, which can then be tested and exported as FZPZ files. This should make live easier for part developers. It should *not* be seen as incentive to distribute FZP files, since, without the corresponding SVG files bundled, the part will not work.