fritzing/fritzing-app #3466: Make sure to use release version of libgit2

KjellMorgenstern, 2019-05-09:
In the develop and fix-windows-ci branches, we automatically build the libgit2 on travis and appveyor.
Unexpectedly, git2.dll ends up in build64/Debug folder. Thus, we build a debug version, which we don't need, plus, our scripts are confused about where to find the dll.

Solution might be to add `--config release`
See also
https://stackoverflow.com/questions/49417434/libgit2-is-not-building-in-release-mode-on-windows

KjellMorgenstern, 2019-05-09:
This should also solve https://github.com/fritzing/fritzing-app/issues/3268

ovidiub13, 2019-05-14:
Indeed `cmake --build .` when used on Windows with Visual Studio uses the `Debug` target by default and ignores the `CMAKE_BUILD_TYPE` value. It's something to do with how Visual Studio works. I've hit this in the past, and #3473 is the only fix for it.

ovidiub13, 2019-05-14:
#3268 needs a documentation update, and we plan to do that after we streamline the build process. IMO we can close #3268 and open a new ticket to "don't forget" to update the build docs.

commit 77af8f79 (2019-05-14): Get libgit from release build path

Fixes #3466
commit 0edb00e0 (2019-05-15): Get libgit from release build path

Fixes #3466
closed 2019-05-16 by ovidiub13