fritzing/fritzing-app #3481: Many pages of the website are still unavailable

IvanDelsinne, 2019-05-19:
Following #2904, I noticed that many pages are still unavailable, especially the homepage ([`http://fritzing.org/`](http://fritzing.org/)).

[image]

bladykast, 2019-05-19:
#3418 From what I can tell, this project is dead.

https://hackaday.com/2019/02/14/the-future-of-fritzing-is-murky-at-best/

I can't even access the cached fritzing downloads on Google.

KjellMorgenstern, 2019-05-20:
We know about the server problems, it’s to old to withstand noise from the internet. 
Please take part in #3443 if you want to discuss future steps.

KjellMorgenstern, 2019-05-20:
Server was rebooted and is kicking again. Migration is not straightforward, so it needs to hold up a little bit longer.

closed 2019-05-20 by KjellMorgenstern