fritzing/fritzing-app #3485: Ubuntu 16.04 LTS – no parts update

rlatoska, 2019-05-31:
I downloaded the **fritzing-0.9.3b.linux.i386.tar.bz2** from http://fritzing.org and installed it in **/usr/share/fritzing/** on my old 32-bit Notebook. This folder belongs to the user (drwxrwxr-x) – not to the root. Fritzing starts and works OK on the first glance. 

When I try to update the parts though, Fritzing starts to search for new parts, then it says that the folder **/user/share/fritzing/fritzing-parts** has been changed(4) and the parts folder cannot be updated automatically. 

The regeneration of the parts database in Fritzing fails as well with the message that the **temporary file could not be opened**. After restarting Fritzing there is a message, that some 15 parts could not be found. This error message appears every time I start Fritzing since.

Is there any solution to this problem on Ubuntu 16.04. LTS?

HaraldRau, 2019-05-31:
Is this possible, my folder /usr/share has the rights of root

`drwxr-xr-x 295 root root 12288 Mai 22 16:44 share`

Try to unzip and execute Fritzing in the folder /home/user/. Then should be no write error.

rlatoska, 2019-06-01:
Well, it worked indeed! 

First I deleted the folder **/usr/share/fritzing**. Then I deleted also the folder **/home/user/.config/Fritzing**. After unzipping the fritzing-0.9.3b.linux.i386.tar.bz2 into **/home/user/Fritzing** and launching it for the first time, Fritzing updated the parts automatically and the error message is gone.

closed 2019-06-01 by rlatoska