fritzing/fritzing-app #3493: Change CD flow?

ReDetection, 2019-06-29:
Hi there! I'm super excited to see a roadmap and development going, however I have a favor to ask. 

Current continuous delivery flow tags every successful build in the git. Unfortunately, this breaks new github feature "Watch releases" which intended to notify on new releases, so instead in notifies about travis builds. Pretty strange, don't you think? :) We are only interested in new actual releases, not automated builds.

Not sure what to suggest though. I assume you tag it to attach artifacts. But why to have them in github, when you can keep it in the travis?

KjellMorgenstern, 2019-06-29:
It is not possible to store anything on travis, virtual machines a created from scratch and deleted after the job is done. Also, github releases requires a tag. If a release is created from a git hash without a tag, github automatically adds the tag.

KjellMorgenstern, 2019-06-29:
Marking them as prerelease or dev releases might be worth a try. 
https://github.com/travis-ci/dpl/issues/234

The later however makes the build unaccessible to anyone outside the Fritzing members.

KjellMorgenstern, 2019-06-29:
CD builds marked as draft, lets see how that works out:
https://github.com/fritzing/fritzing-app/commit/1333c44ad6d6435526d7c1961ff5e6cb997e03a2

KjellMorgenstern, 2019-06-29:
I think the draft was not published (good), but also, only team members can access it (bad). One major idea about the continuous delivery is to get feedback fast, and enable everyone to do so.

ReDetection, 2019-06-30:
Thanks for a quick response and a test! 

Yep, I totally agree with you – everyone should be able to take a look at CD builds. Can we try also a `prerelease: true` instead of a `draft: true` please? I'm not 100% sure but I think not the tag but a release information, attached to the tag, produces notification from github. Hopefully, prerelease doesn't count (though I didn't find any confirmation in the documentation).

> It is not possible to store anything on travis

I didn't know that. My bad. I've fallen into assumption that all CI services work the same. Sorry.

KjellMorgenstern, 2019-06-30:
In the github UI, the button is labeled "publish", so most certainly there will be a notification. I will keep the draft flag for now, and trigger the publish manually whenever fit. This means that there will still be frequent notifications, but at least not multiple possibly insignificant a day. The only alternative I see would be to host the CD builds ourselves, but that comes at high cost: Blocked until new website is finished, work for adapting the scripts, AWS S3 fees.

closed 2019-07-03 by KjellMorgenstern